RobTillaart/ADS1X15 #14: 4channel continues read

Mplex72, 2020-09-25:
I cannot find a way to get the continues read example to 4 channels (or even dual ads1115)

How to select the channels to readout ?  there is only (ADS.getLastValue());  ?

RobTillaart, 2020-09-25:
Hi, good question
It is possible according to the datasheet 9.4.2.2 section by writing a new configuration after starting conversion but before the result is ready.

If time permits i will investigate this tomorrow as it is a good question and it would be interesting to know.

Mplex72, 2020-09-25:
Would have thought , when asked for the last result , it would dump the whole register sequentially ?

RobTillaart, 2020-09-26:
> Would have thought , when asked for the last result , it would dump the whole register sequentially ?

There is only one ADC in the device so no 4 register places for 4 different Analog INports (AINx)

I have written a sketch that runs in continuous mode, uses the ALERT/RDY pin to catch the conversion ready by means of an interrupt routine which sets a flag. It stores the value in an array and request the conversion for the next pin.

Did some tests and it seems to work well, please give it a try.

```cpp
//
//    FILE: ADS_continuous4.ino
//  AUTHOR: Rob.Tillaart
// VERSION: 0.1.0
// PURPOSE: read multiple analog inputs continuously
//          interrupt driven to catch all conversions.
//

// test
// connect multiple potmeters
//
// GND ---[   x   ]------ 5V
//            |
//
// measure at x  - connect to AIN0..4.
//
// for the test it is good to have AIN3 connected to 5V and AIN4 to GND
// so one can see these as references in the output.
//

#include "ADS1X15.h"

// choose you sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);
ADS1115 ADS(0x48);

volatile bool RDY = false;
uint8_t nextPin = 1;
uint8_t lastPin = 0;
int16_t val[4] = { 0, 0, 0, 0 };

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);

  ADS.begin();
  ADS.setGain(0);        // 6.144 volt
  ADS.setDataRate(7);    // slow

  // SET ALERT RDY PIN
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorThresholdLow(0x0000);
  ADS.setComparatorQueConvert(0);

  // SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);

  ADS.setMode(0);        // continuous mode
  ADS.readADC(lastPin);  // trigger first read
  ADS.readADC(nextPin);  // set config to next pin.
}

void loop()
{
  if (RDY)
  {
    // save the last value
    val[lastPin] = ADS.getLastValue();
    // request next pin
    nextPin++;
    if (nextPin >= 4) nextPin = 0;
    ADS.readADC(nextPin);
    lastPin = nextPin;
    RDY = false;
  }

  for (int i = 0; i < 4; i++)
  {
    Serial.print(val[i]);
    Serial.print('\t');
  }
  Serial.println();
  delay(100);
}

void adsReady()
{
  RDY = true;
}

// -- END OF FILE --
```

RobTillaart, 2020-09-26:
The first 4 reads are not always correct, that is because the sensor does not reset when the UNO does. 
I'll have a look at that later, when solved this sketch will be added to the examples. 

After all sensors have been read once they are correct - look at the output in the serial plotter.

Mplex72, 2020-09-26:
That,s perfect , thanks.
It is starting now after only 1 read.
I can experiment on this basis now , I could not find this solution .
The "problem" is solved now .
Klass

RobTillaart, 2020-09-26:
Testrun of 8 hours was stable, showed no new problems here, 

latest variation - put the handling of a new conversion in a function so it can be called at multiple places in loop()

Note now it checks 0,1,2,3 in a round robin fashion, any other pattern could be used here e.g. 0,1,0,2,0,3,0 etc 
which would give channel 0 3x as much priority than the others...

```cpp
//
//    FILE: ADS_continuous4.ino
//  AUTHOR: Rob.Tillaart
// VERSION: 0.1.1
// PURPOSE: read multiple analog inputs continuously
//          interrupt driven to catch all conversions.
//

// test
// connect multiple potmeters
//
// GND ---[   x   ]------ 5V
//            |
//
// measure at x  - connect to AIN0..4.
//
// for the test it is good to have AIN3 connected to 5V and AIN4 to GND
// so one can see these as references in the output.
//

#include "ADS1X15.h"

// choose you sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);
ADS1115 ADS(0x48);

volatile bool RDY = false;
uint8_t channel = 0;
int16_t val[4] = { 0, 0, 0, 0 };

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);

  ADS.begin();
  ADS.setGain(0);        // 6.144 volt
  ADS.setDataRate(7);    // slow

  // SET ALERT RDY PIN
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorThresholdLow(0x0000);
  ADS.setComparatorQueConvert(0);

  // SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);

  ADS.setMode(0);        // continuous mode
  ADS.readADC(channel);     // trigger first read
}

void loop()
{
  handleConversion();

  for (int i = 0; i < 4; i++)
  {
    Serial.print(val[i]);
    Serial.print('\t');
    handleConversion();
  }
  Serial.println();
  delay(100);
}

void adsReady()
{
  RDY = true;
}

void handleConversion()
{
  if (RDY)
  {
    // save the last value
    val[channel] = ADS.getLastValue();
    // request next channel
    channel++;
    if (channel >= 4) channel = 0;
    ADS.readADC(channel);
    RDY = false;
  }
}
// -- END OF FILE --
```

Mplex72, 2020-09-26:
Impressive , looks good and working perfect and stable.
quick look shows pretty "noise free"  signal , just as it should be.
No way to use 2 x 4 channels) this  because  of adsReady and the handle conversion can not be assigned to a specific board ?
(ADS1 , ADS2 etc.) ?

RobTillaart, 2020-09-27:
Of course it can be done as long as processing power of UNO (assumption) is not exceeded.

You can use hardware interrupt 1 on pin 3 to have a 2nd RDY flag and use the same handleConversion function to handle the 2nd RDY flag. A quite identical if block would do the trick. . 

As an ads1115 just uses ~1.1 millisecond per conversion (fastest mode) the load would be 2000 interrupts per second which is a decent load. Depending on what processing the samples need you may need a faster or second processor.

RobTillaart, 2020-09-27:
You can also use pin change interrupts if you need even more channels. But these are slower than the hardware interrupts. 
There are other strategies possible like multiplexing the IRQ pin.

The main questions are:
- How many channels do you need?
- How many samples per second per channel do you need?
- how much time takes processing of a sample?

These 3 numbers determine the limits of the system.

RobTillaart, 2020-09-27:
8 channel version with 2 ADS1115 - only tested with one device as I do not have more spare   

Not too different from the 4 channel version, mostly just doubling code / vars.

You could cleanup things by making a function to initialize and start a single ADS

**bool setupADS(ADS, channel, ....)**

```cpp
//
//    FILE: ADS_continuous8.ino
//  AUTHOR: Rob.Tillaart
// VERSION: 0.1.0
// PURPOSE: read multiple analog inputs continuously
//          interrupt driven to catch all conversions.
//

// test
// connect multiple potmeters to 2 ADS1115
//
// GND ---[   x   ]------ 5V
//            |
//
// measure at x  - connect to AIN0..4.
//
// for the test it is good to have AIN3 connected to 5V and AIN4 to GND
// so one can see these as references in the output.
//

#include "ADS1X15.h"

// adjust addresses if needed
ADS1115 ADS_1(0x48);
ADS1115 ADS_2(0x49);

volatile bool RDY_1 = false;
volatile bool RDY_2 = false;
uint8_t channel_1 = 0;
uint8_t channel_2 = 0;
int16_t val[8] = { 0, 0, 0, 0, 0, 0, 0, 0 };

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  // SETUP FIRST ADS1115
  ADS_1.begin();
  ADS_1.setGain(0);        // 6.144 volt
  ADS_1.setDataRate(7);

  // SET ALERT RDY PIN
  ADS_1.setComparatorThresholdHigh(0x8000);
  ADS_1.setComparatorThresholdLow(0x0000);
  ADS_1.setComparatorQueConvert(0);

  // SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady_1, RISING);

  ADS_1.setMode(0);          // continuous mode
  ADS_1.readADC(channel_1);  // trigger first read

  // SETUP SECOND ADS1115
  ADS_2.begin();
  ADS_2.setGain(0);        // 6.144 volt
  ADS_2.setDataRate(7);

  // SET ALERT RDY PIN
  ADS_2.setComparatorThresholdHigh(0x8000);
  ADS_2.setComparatorThresholdLow(0x0000);
  ADS_2.setComparatorQueConvert(0);

  // SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(3, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(3), adsReady_2, RISING);

  ADS_2.setMode(0);          // continuous mode
  ADS_2.readADC(channel_2);  // trigger first read
}

void loop()
{
  handleConversion();

  for (int i = 0; i < 8; i++)
  {
    Serial.print(val[i]);
    Serial.print('\t');
    handleConversion();
  }
  Serial.println();
  delay(100);
}

// catch interrupt and set flag
void adsReady_1()
{
  RDY_1 = true;
}

void adsReady_2()
{
  RDY_2 = true;
}

// handle conversions that are ready
void handleConversion()
{
  if (RDY_1)
  {
    // save the last value
    val[channel_1] = ADS_1.getLastValue();
    // request next channel
    channel_1++;
    if (channel_1 >= 4) channel_1 = 0;
    ADS_1.readADC(channel_1);
    RDY_1 = false;
  }
  if (RDY_2)
  {
    // save the last value
    val[4 + channel_2] = ADS_2.getLastValue();
    // request next channel
    channel_2++;
    if (channel_2 >= 4) channel_2 = 0;
    ADS_2.readADC(channel_2);
    RDY_2 = false;
  }
}

// -- END OF FILE --
```

Mplex72, 2020-09-27:
My "wish list" is as follows,
 At least 8 channels, preferably 12, of which I want to log at least 100 samples per second.
processing to CAN takes a few milliseconds. 
 I thought I would use it as a background process where a timer determines that it will be read out again.
Then it doesn't use much processor time, does it?
I also read some max31855 channels.

Mplex72, 2020-09-27:
Just as the latest code  , but the READY  check determined by a timer.
" every 100 > read ads1115A"  for pressure sensor
" every 500 > read ads1115B " for slower temperature sensors.
I thougth that with a slow 250Sps or even slow as 128Sps, it would be usable without smoothing after the reads

Mplex72, 2020-09-27:
btw. I use the timer lib from Michael Contrem  , it is very easy to use for a hobbyist like me

RobTillaart, 2020-09-27:
100 samples per channel per second means that you do not need the ADS to run on full speed, 
a data rate of 6 gives 475 samples per second for 4 channels ~ 100 samples per channel per second.

12 channels means 3x 475 interrupts per second ~ 1500 interrupts per second => max 660 usec per sample to handle.
This part should be no problem.

For 3 or more ADS115's you need another way to handle the interrupts when a conversion is ready.
Check - https://github.com/GreyGnome/EnableInterrupt - do not know if it is fast enough
(worked on it long time ago - https://forum.arduino.cc/index.php?topic=87195.0)

RobTillaart, 2020-09-27:
> btw. I use the timer lib from Michael Contrem , it is very easy to use for a hobbyist like me

link?

Mplex72, 2020-09-27:
This one,
https://github.com/contrem/arduino-timer
I have based my code on the blink example

RobTillaart, 2020-09-27:
Ok, think the problem is now beyond this library so I will close the issue.

closed 2020-09-27 by RobTillaart

Mplex72, 2020-09-27:
Yes , you did a perfect job helping me out.
Now I have to do something myself, 
Thanks , Klass

RobTillaart, 2020-09-27:
merged samples in the library as they are not that bad ;)

Success!

RobTillaart, 2020-09-27:
> Just as the latest code , but the READY check determined by a timer.
> " every 100 > read ads1115A" for pressure sensor
> " every 500 > read ads1115B " for slower temperature sensors.
> I thougth that with a slow 250Sps or even slow as 128Sps, it would be usable without smoothing after the reads

Missed this one,
yes you can set the sample rate per ADS, thereby minimizing the number of interrupts. 
depends completely on your requirements and process to monitor.

If heat is supplied by a laser you might need 1000 SPS, while monitoring a fridge 1 SPS is OK.

Note:
to read every channel of an ADS you need 4 reads ==> so the SPS should be 475 to get at least 100 samples per second per channel