RobTillaart/ADS1X15 #18: ADS1115 resolution question

thefreakyZ, 2021-02-02:
Hi there! I’m working on a project and using this Arduino library. I’m using a differential sensor and saw that I cannot see some of the data between readings. Is there anything I could add using Arduino code to see the full resolution? I saw that there was a mention of potential future work on the differential side of things, and was wondering if this referring to resolution. Thanks!

RobTillaart, 2021-02-02:
Hi,

Your description has too little information to understand what you are asking ?

- which ADC are you using 1015 or the 1115?
- which sensor is connected to it
- which version of the library are you referring to?
- what data do you expect **between** readings?
- how often do you make a reading?
- have you minimal sample code that shows the problem?
- the library supports the differential reading, so not clear what you are missing?
- you cannot set the resolution as far as I know my library as this is given by the ADC used. You can decrease and even increase  resolution in your code within certain limits by oversampling.
- what is the goal of your project? what do you try to accomplish? WHat is the bigger picture?

A lot of questions, but they all popped up :) 

Have to do other things now, but back online later today.

thefreakyZ, 2021-02-02:
Hi there! Sorry for the lack of information, I'm new to github and still trying to understand how to go about things.

To make things worse, I just realized I was looking at the wrong library. I'm actually using a different one. I'm so sorry to waste your time. Have a good day!

closed 2021-02-02 by thefreakyZ

RobTillaart, 2021-02-02:
No problem, can happen...
Which lib did you use?