RobTillaart/ADS1X15 #19: ALRT Pin number?

kritch83, 2021-03-26:
This really isn't a issue, but I can't find what pin on the MCU the library looks at for reading the data ready state/pin.  Am I just missing it?  Is there a way to set the pin it watches?  Thanks

RobTillaart, 2021-03-26:
@kritch83 
Good question. 
The library has no way to set an interrupt handler for the ALERT/RDY pin. 
One has to program the interrupt handler outside the library.  

A simple example how the interrupts can be handled =>  ADS_continuous_4_channel.ino 
The ALERT pin is connected to pin2 which is a hardware interrupt on the Arduino UNO and if triggered it sets the RDY flag which is checked and handled somewhere in ```handleConversion``` which is called pretty often. 

The flag RDY should have been guarded with ```nointerrupts()   interrupt()``` 
but as the ISR just sets a one byte flag it works well without guards. 

Problem with interrupts in libraries is that I did not find a way to implement them in a clean, understandable and portable way. 

Does this answer your question ?

kritch83, 2021-03-27:
Yes that does.  IDK how I missed that example.  Thought I looked thru them all.  Thank you!

closed 2021-03-27 by kritch83