RobTillaart/ADS1X15 #20: Allow specifying I2C bus

MaffooClock, 2021-03-29:
This library hardcodes the `Wire` bus, so it's impossible to use this with sensors on any other I2C bus that may be available (`Wire1`, `Wire2`, etc.).

I may see if I can handle this and submit a PR...

RobTillaart, 2021-03-29:
Thanks for reporting, it is on my todo list, I prefer to use a similar construct in all my libs.
So it should be a bit like - https://github.com/RobTillaart/PCF8574/blob/master/PCF8574.h 

I have some time so you might expect a development stream in an hour or so.

MaffooClock, 2021-03-29:
> I have some time so you might expect a development stream in an hour or so.

Cool!  I'd already started on it (seems pretty simple to do) since my project cannot continue without this, but I'll hold off for now since you're on it.

RobTillaart, 2021-03-29:
develop branch is uploaded, tests are running

MaffooClock, 2021-03-29:
Perfect!

RobTillaart, 2021-03-29:
Which platform? (for reference)

I will merge the branch, thanks for testing!

MaffooClock, 2021-03-29:
> Which platform? (for reference)
Teensy 4.1

RobTillaart, 2021-03-29:
OK, if there are more issues, just let me know!

closed 2021-03-29 by RobTillaart