RobTillaart/ADS1X15 #26: Documentation clarification. In latch mode, alert pin goes high only after readADC?

alicedb2, 2021-07-10:
I _think_ this might just be and issue about clarifying the documentation.

**Here's the setup:**
- I'm using the traditional + latch mode,
- Switch connected to A0. When the switch is pressed A0 goes HIGH,
- ALERT pin is connected to D2,
- When A0 goes HIGH I want the ALERT pin to trigger an interrupt attached to D2,
- When the interrupt is called I set a flag so that something happens in loop(), in this case print "foo".

**Issue?**
- In the code below, when `ADS.readADC(0)` is called before `if (read_ADC){...}` everything goes as intended,
- If it is called within, i.e. `if (read_ADC) { ADS.readADC(0); ...}` then what's inside the `if` then the ALERT pin doesn't go HIGH, the interrupt on D2 is not executed so `do_thing` remains `false`, and nothing happens.
- So it appears that you have to `readADC()` to both make ALERT go HIGH and immediately low again? Why call it LATCH mode?

I thought that the order of event was
- A0 goes HIGH, triggering an interrupt and ALERT goes HIGH, regardless of if you're in LATCH mode or not.
- **After** ALERT goes HIGH, you need to `readADC()` so that it goes low again.

Here's the code in question.

```
#include <avr/interrupt.h>
#include <Wire.h>
#include <ADS1X15.h>

ADS1115 ADS(0x48);

volatile bool do_thing = false;

void setup()
{
  pinMode(2, INPUT);
  attachInterrupt(digitalPinToInterrupt(2), foo, RISING);

  Serial.begin(115200);

  Wire.begin();
  Wire.setClock(400000);

  ADS.begin();
  ADS.setGain(0);
  ADS.setComparatorMode(0);
  ADS.setComparatorPolarity(0);
  ADS.setComparatorLatch(1);
  ADS.setComparatorQueConvert(0);
  ADS.setComparatorThresholdLow(2000);
  ADS.setComparatorThresholdHigh(30000);

}

void loop()
{
  int16_t val_0 = ADS.readADC(0);   // code works as intended if here

  if (do_thing) {
    // int16_t val_0 = ADS.readADC(0);   // code doesn't work as intended if here

    Serial.println("foo");
    do_thing = false;
  }
}

void foo() {
  do_thing = true;
}```

RobTillaart, 2021-07-11:
(added cpp after the backquotes of code section of your post to get syntax highlighting)

Thank you for reporting the issue, it will take some days before I can recreate your setup.
I need to study the datasheet again to get into the details. So a quick response from my head.

**theory**

When **LATCH** is set, the **ALERT** will not send a  pulse but it will keep the line **HIGH** until a new read is started.  
The line will be reset after a new **readADC(...)** if I recall correctly.

**why latching**

The latching allows one to poll the pin instead of using an interrupt. 
Latching also allows one to reexamine the state of the pin, during or after the interrupt. 
It can be useful in a multithreaded environment, although this can lead to complex code / behavior.
A drawback of latching is that only a new readADC() will reset the line, and sometimes you might not want to do that.
Can you explain why set the latching? 

**First analysis**

Looking at your code I expect your code needs an initial read to start the chain of reads working as you want.  
In your code you use the first line in **loop()** for that to get it working. 
By moving this initial read to setup() you get the following code

Can you try this
```cpp
// TESTCODE 1
#include <avr/interrupt.h>
#include <Wire.h>
#include <ADS1X15.h>

ADS1115 ADS(0x48);

volatile bool do_thing = false;

void setup()
{
  pinMode(2, INPUT);
  attachInterrupt(digitalPinToInterrupt(2), foo, RISING);

  Serial.begin(115200);

  Wire.begin();
  Wire.setClock(400000);

  ADS.begin();
  ADS.setGain(0);
  ADS.setComparatorMode(0);
  ADS.setComparatorPolarity(0);
  ADS.setComparatorLatch(1);
  ADS.setComparatorQueConvert(0);
  ADS.setComparatorThresholdLow(2000);
  ADS.setComparatorThresholdHigh(30000);

  ADS.readADC(0);   // start the chain of conversions 
}

void loop()
{
  if (do_thing) {
    int16_t val_0 = ADS.readADC(0);  

    Serial.print(micros());   // just added to get insight in the timing of the ADC.
    Serial.println("\tfoo");
    do_thing = false;
  }
}

void foo() {
  do_thing = true;
}
```

----

Another (less nice) way to start the chain of reads is to set **do_thing** initially to true.
Simulates an initial interrupt. This should work, however if this is acceptable depends on 
your code. Does it needs an actual value from the ADC (assume it is) after an interrupt?

```cpp
// TESTCODE 2
#include <avr/interrupt.h>
#include <Wire.h>
#include <ADS1X15.h>

ADS1115 ADS(0x48);

volatile bool do_thing = true;     // will trigger the first read 

void setup()
{
  pinMode(2, INPUT);
  attachInterrupt(digitalPinToInterrupt(2), foo, RISING);

  Serial.begin(115200);

  Wire.begin();
  Wire.setClock(400000);

  ADS.begin();
  ADS.setGain(0);
  ADS.setComparatorMode(0);
  ADS.setComparatorPolarity(0);
  ADS.setComparatorLatch(1);
  ADS.setComparatorQueConvert(0);
  ADS.setComparatorThresholdLow(2000);
  ADS.setComparatorThresholdHigh(30000);
}

void loop()
{
  if (do_thing) {
    int16_t val_0 = ADS.readADC(0);  

    Serial.print(micros());
    Serial.println("\tfoo");
    do_thing = false;
  }
}

void foo() {
  do_thing = true;
}
```

RobTillaart, 2021-07-11:
Just a thought, It might be possible that you could get the triggering to work by setting the **continuous** mode.
Never tried

you should add this line to the end of setup().
```cpp
setMode(0);    // set continuous mode
```

RobTillaart, 2021-07-31:
@alicedb2 
Is this issue solved now / answered ?

alicedb2, 2021-08-09:
``setMode(0)`` appears to have helped but now the ALERT pin goes high when A0 is high and goes low when A0 is low even though LATCH mode is on.

I'm using the code below. The expected behavior is
- In the beginning ``do_thing=false``,
- I press and release a switch and A0 goes high and a fraction of a second later goes low.
- The moment I depressed the switch the ALERT pin goes high **and latches** even after I release the switch,
- The ``CHANGE`` interrupt on D2 triggers and ``do_thing=true``,
- ``loop()`` start printing ``micros()``,
- Three seconds later ``readADC(0)`` makes the ALERT pin go low,
- The ``CHANGE`` interrupt on D2 triggers again and ``do_thing=false``,
- ``loop()`` stops printing ``micros()``.

The actual behavior is
- In the beginning ``do_thing=false``,
- I depress a switch and A0 goes high and a fraction of a seconds later I release the switch and A0 goes low
- The ALERT pin goes high and ``do_thing=true`` when the switch is pressed,
- Interrupt is triggered on rising and ``loop()`` prints ``micros()`` for a fraction of a second,
- Then ALERT pin goes low (**ALERT didn't latch!**) in sync with when I release the switch, 
- Interrupt is triggered on falling and changes back to ``do_thing=false``
- ``loop()`` stops printing ``micros()``

```
// TESTCODE 3

#include <avr/interrupt.h>
#include <Wire.h>
#include <ADS1X15.h>

ADS1115 ADS(0x48);

volatile bool do_thing = false;
volatile bool readadc = false;
bool wait = false;
long int last_do = 0;

void setup()
{
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), foo, CHANGE);

  Serial.begin(115200);

  Wire.begin();
  Wire.setClock(400000);

  ADS.begin();
  ADS.setGain(0);
  ADS.setMode(0);
  ADS.setComparatorMode(0);
  ADS.setComparatorPolarity(0);
  ADS.setComparatorLatch(1);
  ADS.setComparatorQueConvert(0);
  ADS.setComparatorThresholdLow(2000);
  ADS.setComparatorThresholdHigh(20000);

//  ADS.readADC(0);   // start the chain of conversions
}

void loop()
{
  if (do_thing) {
    last_do = millis();
    wait = true;
    Serial.println(micros());
  }

  if (millis() - last_do > 3000 && wait) {
    Serial.println(ADS.readADC(0));
    wait = false;
  }

}

void foo() {
  do_thing = !do_thing;
}

RobTillaart, 2021-08-09:
The latch pin goes **LOW** for longer time only in **comparator** mode. 
Typical use is putting the ADC in **continuous** mode and trigger if an unwanted value is reached.
See picture from the datasheet. 

[image]

RobTillaart, 2021-08-09:
If the comparator is in **window** mode, the picture is slightly different:
Latching, it waits for a specific command
Non latching, you get a pulse when a threshold is passed

[image]

RobTillaart, 2021-08-09:
From your code

```cpp
  ADS.setComparatorMode(0);       // traditional
  ADS.setComparatorPolarity(0);
  ADS.setComparatorLatch(1);        // set latching mode
```

So your settings are OK. However the flags are only set after a **_requestADC()**.
**_requestADC()** is called (a.o) by **_readADC()** which is called by **readADC()**

And that line is commented in your sketch 
```cpp
//  ADS.readADC(0);   // start the chain of conversions
```
So it looks like you are setting the flag correctly, but it is **not** forwarded to the hardware 
That is an library issue and yes that should be made explicit in the documentation. (action point for me)

Can you try to uncomment the line mentioned ? 
That should forward the value of the latch flag to the hardware.
Please let me know the results.

If this not works I have to do recreate a test with hardware.

RobTillaart, 2021-08-09:
I updated the **readme.md** file in the master branch 
- reflect that updating the flags need a **readADC()**
- added type information about the methods / parameters
- fixed some typos

RobTillaart, 2021-08-29:
Is the question answered? if so the issue can be closed?

RobTillaart, 2021-09-05:
If the issue still exists feel free to reopen the issue.

closed 2021-09-05 by RobTillaart