RobTillaart/ADS1X15 #32: ADS1115 Data Rate question

sghirsst, 2021-10-20:
Hello Rob, firstly thank you for the library and your effort.
I have an issue regarding the data rate in the library, when I set the data rate to a slowest rate value for example 4 it does not respond with 128 samples/second, it gives around 250~300.
Also when I set the rate to 7 (860 samples/second) it gives around 817 sps, please find the below the snippet of code I'm using, I need to know the source of the problem, is it a hardware issue or software, thank you in advance.

Environment:
- ESP32
- ADS1115

Arduino Code:
```
#include "ADS1X15.h"

ADS1115 ADS(0x48);
float f = 0;
volatile bool convReady = false;

void convReadyAlert(){
   convReady = true;
}
void setup() 
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);
  
  ADS.begin(21,22); // SDA, SCL
  ADS.setGain(1);     
  ADS.setDataRate(4);  // DATA RATE
  ADS.setMode(0); 
  // set the thresholds to Trigger RDY pin
  ADS.setComparatorThresholdLow(0x0000);
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorQueConvert(0);             // enable RDY pin !!
  
  pinMode(23, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(23), convReadyAlert, RISING);
  ADS.requestADC(0);
}

void loop() 
{
  if (convReady)
  { 
    int16_t val_0 = ADS.getValue();
    Serial.print("\Value: ");
    Serial.println(val_0 * f, 3);
    convReady = false;
    ADS.requestADC(0);
    
  }
}

```

sghirsst, 2021-10-20:
For the data rate 6 it gives 450~475 SPS which is also correct taking into consideration the variation of +- 10%

RobTillaart, 2021-10-20:
If you set the datasheet to 0 I expect you will also get ~275 sps.

IN the datasheet there is a section **DUTY CYCLING FOR LOW POWER**
If I understand it right the internal ADC does not lower the sampling rate but does oversampling at lower speeds.
Check if I can get that confirmed.

RobTillaart, 2021-11-03:
@sghirsst 
Is this issue still a problem?

sghirsst, 2021-11-03:
@RobTillaart , no the problem is solved and everything is working fine, thank you very much.

closed 2021-11-03 by RobTillaart