RobTillaart/ADS1X15 #36: Not working with Nano Every

80stech, 2022-01-21:
Hi, this looks like a really good library :) but I can't get anything to compile for the Nano Every.  AVR dude seems to have problems with the get clock (what you have listed as a prototype) and set clock functions and a few other things I think related to setting clocks for the I2C bus. I tried to copy the error messages but I not able to. Is there maybe something I can change in the header file to get it working?  Thanks! :)  Rainer S.

RobTillaart, 2022-01-21:
Thanks for reporting the issue.
I am not familiar with the Nano Every. The getClock is indeed experimental as it is not a standard implement  feature on all boards. Tomorrow I will do a first investigation (here it is 3.30 AM now) if time permits. Otherwise it will be Saturday. 
Do you have a link to the board?

RobTillaart, 2022-01-21:
Found the board, so link is not needed.

80stech, 2022-01-21:
OK, Thanks for looking into this! :)

Cheers,
Rainer S.

RobTillaart, 2022-01-21:
- installed NANO  every board
- compiled the minimum example and got errors

1) TWBR => this is an Arduino UNO low level I2C register.  I understand this issue,
2) _wire->requestFrom(address, (uint8_t) 2);  error =>  casting error, also understood 

Both seems easy fixable however I do not want to introduce a problem for the existing platforms.
I will create a develop branch in which this I try to solve the issue.
Can you do some testing with the real hardware?

RobTillaart, 2022-01-21:
Problem 1 is related to this issue - https://github.com/arduino/Arduino/issues/11457

RobTillaart, 2022-01-21:
Please verify if develop branch works for you - https://github.com/RobTillaart/ADS1X15/tree/develop

It compiles and I did not found regression of other platforms.

80stech, 2022-01-21:
Sorry I just woke up and I have a dentist appointment this morning, but I'll try this out ASAP! :)

RobTillaart, 2022-01-21:
No problem, take your time 
Good luck at the dentist.

80stech, 2022-01-21:
I tested both minimum and performance sketches with ADS1015 and Nano Every and all is good! :)
Thanks you very much :)
Cheers,
Rainer S.

RobTillaart, 2022-01-21:
Good to hear, 
I will merge and create a new release

closed 2022-01-21 by RobTillaart

RobTillaart, 2022-01-21:
Released 0.3.5 !

commit a05e819c (2022-01-21): fix #36 Support NANO Every (#37)

* fix #36 Support NANO Every