RobTillaart/ADS1X15 #42: ADS1115 reading

ShaneKnightCoder, 2022-06-04:
hello, I'm working on a project where Im using the ads1115 with esp32 .
Im using ESPIDF v4.2 in let me know if you getting feel any update or need to change here.
	// Select configuration register(0x01)
	// AINP = AIN0 reading voltage 
	// Continuous conversion mode, 128 SPS(0xC1, 0x83)

hardware is fine and working I checked in Arduino IDE but here I'm only getting reading 0x5E, 0x5E  

static esp_err_t i2c_master_sensor_test(i2c_port_t i2c_num, uint8_t *data_h, uint8_t *data_l)
{
    int ret;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, 0x90, 0x01);
    i2c_master_write_byte(cmd, 0x01, 0x01);
    i2c_master_write_byte(cmd, 0xC1, 0x01);
    i2c_master_write_byte(cmd, 0x83, 0x00);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    if (ret != ESP_OK) {
        return ret;
    }
    vTaskDelay(30 / portTICK_RATE_MS);
    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, 0x91, 0x01);
    i2c_master_write_byte(cmd, 0x00, 0x01);
    i2c_master_read_byte(cmd, data_h, 0x00);
    i2c_master_read_byte(cmd, data_l, 0x01);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
        printf(" >> data_h: %X\n", *data_h);
        printf(" >> data_l: %X\n", *data_l);
        return ret;
}

RobTillaart, 2022-06-04:
I'm not so familiar with this ESP coding. 
Will try to investigate next week.

RobTillaart, 2022-06-05:
First question that pops up is 
- what is the value of **ret** when you print the two numbers.

RobTillaart, 2022-06-05:
Think the reading does not set the register right.

Try this.

```cpp
    i2c_master_write_byte(cmd, 0x90, 0x01);   //  address + write bit
    i2c_master_write_byte(cmd, 0x00, 0x01);   //  conversion register
    i2c_master_write_byte(cmd, 0x91, 0x01);   //  address + read bit
    i2c_master_read_byte(cmd, data_h, 0x00);   // read byte, ACK
    i2c_master_read_byte(cmd, data_l, 0x01);    // read byte, NACK
```

ShaneKnightCoder, 2022-06-05:
@RobTillaart  Thank you for responding 

Q1: what is the value of ret when you print the two numbers.
Ans. that is 263 and I drive more in deep I think I'm not getting the ACK here:
    i2c_master_write_byte(cmd, 0x91, 0x01);
    i2c_master_write_byte(cmd, 0x00, 0x01);  <<  (not getting ACK)
Q2: what is the device address of your sensor?
Ans. I'm using the ADS1115 ( 0x90 writing 0x91 reading )
Q3: you send "magic numbers" in the command. Comments could explain what the code is expected to do.
   which "magic number"

ShaneKnightCoder, 2022-06-05:
> Think the reading does not set the register right.
> 
> Try this.
> 
> ```c++
>     i2c_master_write_byte(cmd, 0x90, 0x01);    //  address + write bit
>     i2c_master_write_byte(cmd, 0x00, 0x01);    //  conversion register
>     i2c_master_write_byte(cmd, 0x91, 0x01);    //  address + read bit
>     i2c_master_read_byte(cmd, data_h, 0x00);   //  read byte, ACK
>     i2c_master_read_byte(cmd, data_l, 0x01);   //  read byte, NACK
> ```

Yes exactly

RobTillaart, 2022-06-05:
> which "magic number" ???

All raw numbers are "magic", when you send 0x90 or 0x91 that is just a number

compare

```cpp
    i2c_master_write_byte(cmd, 0x90, 0x01);    //  address + write bit
    i2c_master_write_byte(cmd, 0x00, 0x01);    //  conversion register
    i2c_master_write_byte(cmd, 0x91, 0x01);    //  address + read bit
    i2c_master_read_byte(cmd, data_h, 0x00);   //  read byte, ACK
    i2c_master_read_byte(cmd, data_l, 0x01);   //  read byte, NACK
```
to

```cpp
    i2c_master_write_byte(cmd, ADDRESS | WRITE_FLAG, 0x01);
    i2c_master_write_byte(cmd, CONV_REGISTER, 0x01);
    i2c_master_write_byte(cmd, ADDRESS | READ_FLAG, 0x01);
    i2c_master_read_byte(cmd, data_h, ACK_FLAG);
    i2c_master_read_byte(cmd, data_l, NACK_FLAG);
```

With a number of defines the code makes more sense without comments

ShaneKnightCoder, 2022-06-05:
Thank you helping @RobTillaart .. I was working like same as you suggested thank I changed them to direct value I thought may be there is an issue but Yaaa getting the same results.

RobTillaart, 2022-06-05:
Do you mean that the code now works as it should?

ShaneKnightCoder, 2022-06-05:
> Do you mean that the code now works as it should?

I mean still not getting the result as I'm looking. issue is still there

RobTillaart, 2022-06-05:
And if you use the default configuration?   0x8583   (from datasheet page 18)
Then the 4 lines of the ADS1115 should work as a simple ADC

```cpp
static esp_err_t i2c_master_sensor_test(i2c_port_t i2c_num, uint8_t *data_h, uint8_t *data_l)
{
    int ret;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, 0x90, 0x01);
    i2c_master_write_byte(cmd, 0x01, 0x01);
    i2c_master_write_byte(cmd, 0x85, 0x01);  // <<<<
    i2c_master_write_byte(cmd, 0x83, 0x00);  // <<<<
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    if (ret != ESP_OK) {
        return ret;
    }
...
```

ShaneKnightCoder, 2022-06-05:
Totally agree with you buddy. I this that part is OK. I'm having issue while reading.
    i2c_master_write_byte(cmd, ADDRESS | WRITE_FLAG, 0x01);
    i2c_master_write_byte(cmd, CONV_REGISTER, 0x01);     << after that line I'm not getting ACK from salve side 

and the value are the same as I config while writing.

and BTW  

> And if you use the default configuration? 0x8583 (from datasheet page 18)

this is on page 28 in datasheet

RobTillaart, 2022-06-06:
I looked in an 2009 version. In the latest I have it is indeed page 28. (register default content did not change)
Latest version: **SBAS563D –DECEMBER 2011–REVISED JANUARY 2018**
There might be an update since 2018, did not check.

----
wrt the problem at hand, there might be a STOP missing

[image]
```cpp
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ADDRESS | WRITE_FLAG, 0x01);
    i2c_master_write_byte(cmd, CONV_REGISTER, 0x01);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ADDRESS | READ_FLAG, 0x01);
    i2c_master_read_byte(cmd, data_h, ACK_FLAG);
    i2c_master_read_byte(cmd, data_l, NACK_FLAG);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
```

ShaneKnightCoder, 2022-06-06:
@RobTillaart  Yes I thing you get it right .
That is working for me now 
Thank you for support and response

closed 2022-06-06 by ShaneKnightCoder

RobTillaart, 2022-06-06:
You are welcome.