RobTillaart/ADS1X15 #45: ADS1115 on RP2040 Wire, 0x4B is busy

f5exo, 2022-08-13:
Hi,
I have just discovered a problem in the use of the ADS1115 connected Wire0 of the RP2040.
The 3 addresses work correctly except 0x4B which is Busy.
Initially the assembly and the program was functional for the 4 addresses with Arduino Mega.
With the Raspberry Pico everything works correctly, except the address 0x4B which is seen Busy at initialization.
For information on the I2C bus there is also an RTC DS3231 which is on Wire1, but was also present on the Arduino version.
With the Raspberry Pico version I tested the presence of the 3 simultaneous addresses without any problem, only when 0x4B is connected to the assembly, it is not recognized.
I turn in circles.
Thank you for you help.

Erick
Laval in France

RobTillaart, 2022-08-13:
Thanks for filing the issue,

I have no RP2040 so I cannot confirm the issue so I need to think how to tackle this one.

(I see a a lot of bad news from France - fires, canicule(?) - also in your neighbourhood?)

RobTillaart, 2022-08-13:
Do you have an I2C scanner application?
Please disconnect all external devices, do a scan to see if there are "hidden" devices on the board.

if no 0x4B device, connect the ADS1115 as 0x4B and try again.

RobTillaart, 2022-08-13:
Given that the ADS was working with the MEGA (5V) indicates that the ADS itself is OK.

another thought...
Are you using pull up resistors on the SCL / SDA lines? 
if so which value?
As the RP2040 is a 3V3 device, there might be an electric difference, so you might try without pull ups.
(to get address 0x4B the ADDR pin needs to connect to SCL, that's why I ask)

RobTillaart, 2022-08-13:
Found this issue - https://github.com/arduino/ArduinoCore-mbed/issues/448 -
It might or might not be related, at least it is also the SCL line that fails. 

Have you tried switching the devices on Wire0 and Wire1 bus?

f5exo, 2022-08-13:
Thank's for your reply Rob.
I am in the west of France, and I confirm it is very hot.
Yes the ADS is supplied with 3.3V
On arduino it was supplied with 5V
I use the same probes addressable by stap.
which are functional on arduino.

Here is simple test file:
only the 3 addresses are seen, alone or together.
I am looking for an i2c scanner in C not in python.

//-- ADS1115 16 bits --
//https://github.com/RobTillaart/ADS1X15
#include "ADS1X15.h"
// ADS1115 address
ADS1115 ADC_A(0x48);
ADS1115 ADC_B(0x49);
ADS1115 ADC_C(0x4A);
ADS1115 ADC_D(0x4B);

// -- module RTC --
#include "RTClib.h"
#define TIME_24_HOUR true
RTC_DS3231 rtc;

// -- Librairies TFT 3.5  --
#include "TFT_eSPI.h"
TFT_eSPI tft = TFT_eSPI();

bool sonde_A = false;
bool sonde_B = false;
bool sonde_C = false;
bool sonde_D = false;

// Gain value
int G_A = 2;
int G_B = 2;
int G_C = 2;
int G_D = 2;

//----------------SETUP--------------------------------------
void setup() {

  // vitesse d'envoi  pour l'USB
  Serial.begin(57600);  // debug
  //inputString.reserve(200);
  Serial1.begin(57600);
  //-------------------------------------------------------------
  // -----------Module horloge temps reel -----------------------
  //-------------------------------------------------------------
  // mise a jour horloge RTC
  // **** verifier sir on est bien en wire1 ****
  if (!rtc.begin(&Wire1)) {
    Serial.println(F("Couldn't find RTC"));
    Serial.flush();
    abort();
  }
 
  // initialisation Ecran
  tft.init();
  tft.setRotation(3);
  //tft.fillScreen(0x0000);
  tft.fillScreen(TFT_BLACK);

  // -- probe presence test --
  test_Probe();
  ADS1115_Init();
}

//-----------------------------------------------------------
//     ** Probe presence test **
//-----------------------------------------------------------
void test_Probe()
{  
  tft.fillScreen(0x0000);

  // Test if the probes respond

  if (ADC_A.begin()) sonde_A = true;
  if (ADC_A.isConnected())tft.println(F("Sonde A is connected"));
  if (ADC_A.isBusy())tft.println(F("Sonde A is Busy"));
  delayMicroseconds(200);

  if (ADC_B.begin()) sonde_B = true;
  if (ADC_B.isConnected())tft.println(F("Sonde B is connected"));
  if (ADC_B.isBusy()) tft.println(F("Sonde B is Busy"));
  delayMicroseconds(200);

  if (ADC_C.begin()) sonde_C = true;
  if (ADC_C.isConnected()) tft.println(F("Sonde C is connected"));
  if (ADC_C.isBusy()) tft.println(F("Sonde C is Busy"));
  delayMicroseconds(200);

  if (ADC_D.begin()) sonde_D = true;
  if (ADC_D.isConnected()) tft.println(F("Sonde D is connected"));
  if (ADC_D.isBusy()) tft.println(F("Sonde D is Busy"));
  delayMicroseconds(200);

  delay(9900);
}

void ADS1115_Init()
{
  // --  initialisation ADS1115 --
  if (sonde_A == true) {
    ADC_A.begin();
    ADC_A.setGain(G_A);    
    ADC_A.setMode(1); 
    ADC_A.setDataRate(6);  
    ADC_A.readADC(0);
  }
  //delay(40);
  if (sonde_B == true) {
    ADC_B.begin();
    ADC_B.setGain(G_B);
    ADC_B.setMode(1);
    ADC_B.setDataRate(5);
    ADC_B.readADC(0);
  }
  //delay(40);
  if (sonde_C == true) {
    ADC_C.begin();
    ADC_C.setGain(G_C);
    ADC_C.setMode(1);
    ADC_C.setDataRate(6);
    ADC_C.readADC(0);
  }
  //delay(40);
  if (sonde_D == true) {
    ADC_D.begin();
    ADC_D.setGain(G_D);
    ADC_D.setMode(1);
    ADC_D.setDataRate(5);
    ADC_D.readADC(0);
  }
  //delay(40);
  /*
   val 0 Gain de 2/3 (par défaut)  +/- 6.144V  0.1875mV
   1 Gain de 1 +/- 4.096V    0.125mV
   2 Gain de 2 +/- 2.048V    0.0625mV
   4 Gain de 4 +/- 1.024V    0.03125mV
   8 Gain de 8 +/- 0.512V    0.015625mV
   16 Gain de 16  +/- 0.256V  0.0078125mV
  */
}

void loop() {
}

RobTillaart, 2022-08-13:
> I am looking for an i2c scanner in C not in python.

maybe this one - https://github.com/RobTillaart/MultiSpeedI2CScanner

RobTillaart, 2022-08-13:
Your code shows no things that look wrong (as 3 out of 4 work it should be OK)

What is the address of the RTC?

How is the TFT controlled? 
SPI I assume?

f5exo, 2022-08-13:
Are you using pull up resistors on the SCL / SDA lines?
if so which value?
I just looked on the module used (aliexpress) and yes there is a pull up at 10k for these two.

i don't use the 3.3v of Rapsberry , i use 5v with 3.3  external regulator.

As the RP2040 is a 3V3 device, there might be an electric difference, so you might try without pull ups.
(to get address 0x4B the ADDR pin needs to connect to SCL, that's why I ask)

Yes i have a PCB with addressable pins, and i can choose one of four adress.
So I can connect the 4 probes on the same bus with the correct addresses.

TFT is on SPI0

f5exo, 2022-08-13:
Last news
I confirm that with the simplest assembly: RP2040 and ADS1115 only:  the result is identical .
[image]

Serial result.

TEST ADS1115
Sonde A is connected
Sonde A is OK
Sonde B is Busy
Sonde C is Busy
Sonde D is Busy
TEST ADS1115
Sonde A is Busy
Sonde B is Busy
Sonde C is Busy
Sonde D is Busy

f5exo, 2022-08-14:
Hello Rob
Good news
Finally, I added a pull up of 4.7K on the I2C Bus
and now I see each address alone, or all 4 together.

Thank you very much for your help, it helped me a lot.
The best for you.

With the scan I was able to see the address of the RTC: 0X68

RobTillaart, 2022-08-14:
mmm, no clue yet 
Have you tried a smaller pull up resistor e.g. 4K7 or 2K2 or even 1K?
Do you have a scope to see the signals over the I2C bus?

RobTillaart, 2022-08-14:
Good to hear it is solved, you may close the issue .

f5exo, 2022-08-14:
I hear emails before I receive them....hi
Thanks you for all.
Erick at Laval West of France.

closed 2022-08-14 by f5exo

RobTillaart, 2022-08-14:
You're welcome