RobTillaart/ADS1X15 #49: Arduino Micro and ADS1115 - 860SPS (Only on 1 channel ?)

MopheusDG, 2023-03-25:
Hi everyone, I have a question/issue with my ADS1115 and I'd like to know if my conclusion is right or not.

I wrote this small code to test the speed of the ADS1115:

`#include "ADS1X15.h"

// ADS1x15 Definitions
ADS1115 ADS(0x48);

// Timing
unsigned long starttime;
unsigned long endtime ;

void setup() {
  Serial.begin(250000);
  ADS.begin();
  ADS.setGain(0);
  ADS.setDataRate(7);
  ADS.setMode(0);
  ADS.setWireClock(400000);
  if (ADS.isConnected()) ADS.readADC(0);
}

void loop() {
  starttime = micros();
  // Serial.print("Sensor 0: "); Serial.print(ADS.readADC(0));
  Serial.print(" - Sensor 1: ");  Serial.print(ADS.readADC(1));
  // Serial.print(" - Sensor 2: "); Serial.print(ADS.readADC(2));
  endtime = micros();
  Serial.print(" => Tiempo de 1 lectura: "); Serial.print(endtime - starttime);  Serial.print(" us");
  Serial.println();
  delay(1000);
}
`

**Now, when I test this with library delay set on 8, I get: ~8726 us**
#define ADS1115_CONVERSION_DELAY    8

**If I set this to 1, I get: ~1700 us**
#define ADS1115_CONVERSION_DELAY    1

My issue is, I need to read 3 analogs, not 1, as you can see I comment the rest of the readings to check response time PER channel. So, when I add the rest of the channels to this project, the max speed I can get is:

**Now, when I test this with library delay set on 6, I get: ~ 6500 PER CHANNEL**
#define ADS1115_CONVERSION_DELAY    6

**Anything below a delay of 6, the reading get's stuck or star mixing results.**

Conclusion: If I'm not wrong, switching channels on ADS1115 requieres an additional delay and that's what's causing the problem. Am I right ? Is there any way to improve this results ?

Thanks a lot.

RobTillaart, 2023-03-25:
Thanks for the issue, I try to answer asap.

RobTillaart, 2023-03-25:
To maximize the performance you need to run in continuous mode like in the example. 
Have you tried that one?

MopheusDG, 2023-03-25:
ADS.setMode(0); is not continuos mode ?

RobTillaart, 2023-03-25:
From phone, not near my pc, so some half answers.

If I see correctly your code also measures the Serial print statement. Should assign the value to a var and print it outside the time measurement.

Printing takes time too.

Mode(0) is continuous mode,

MopheusDG, 2023-03-25:
I've moved the prints to the end of the loop to save time, but still same performance. Everytime I read 3 channels my timing per channel jumps to ~6500 us. Tried mode 0, mode 1... nothing, both the same result. As soon as I reduce delays on readings and time goes under 6500, the conversion gets stuck and stops reading.

I'm starting to think that with this ADS I can only get 1500 us using 1 channel and not switching to another....

RobTillaart, 2023-03-26:
Spend some time in the code of the library and the datasheet to refresh the mind.

### ADS1115_CONVERSION_DELAY

#define ADS1115_CONVERSION_DELAY 8

Defines the number of milliseconds between the **_requestADC()** and the **getValue**() in the continuous mode.

From the latest code.
```cpp
int16_t ADS1X15::_readADC(uint16_t readmode)
{
  _requestADC(readmode);
  if (_mode == ADS1X15_MODE_SINGLE)
  {
    while ( isBusy() ) yield();   //  wait for conversion; yield for ESP.
  }
  else
  {
    delay(_conversionDelay);      //  TODO needed in continuous mode?
  }
  return getValue();
}
```

There is still a TODO there to investigate this conversionDelay which has low prio as the continuous mode prefers the Async methods.

#### SPS

The max SPS is the maximum number of conversions the chip can make. The 4 channels multiplex one ADC

From datasheet
[image]

#### I2C clock speed

The ADS1x15 support up to 3.4 MHz I2C clock so setting it to the highest possible value supported by your board (and cabling/pull up etc) will gain some time. 
Q: what board do you use?

#### Continuous mode

The advantage of continuous mode is that you do not need to send requests saving (50%?) of the communications, except for an initial request to start the continuous mode.
In the continuous mode you need to watch the RDY pin (conversion ready) for interrupts  (either poll that pin or attach an interrupt) before reading the next value with **getValue()**. In my experience this is the only way to maximize the SPS.
Note: you should not use the **getValue()** inside the interrupt as those should be as short as possible, (however I never tried so it might work). Setting a flag and handle it outside the IRQ is the way to go.

The performance of the continuous mode is defined by the **ADS.setDataRate(7);** function.

In your sketch above (updated for highlighting) you are using **readADC()** which is an explicit blocking function. You should rewrite it to use **getValue()** - see examples.

#### Main question

Your main question is "how should I read 3 channels of the 4 in continuous mode?".  (my interpretation).

Interesting, at least as I never tried it has some challenges. Need to think that over how that could be done.

(to be continued)

RobTillaart, 2023-03-26:
#### 3 channel thoughts

Don't know which is acceptable for you, but here some thoughts.

#### Alternative 1:
The continuous mode will "rotate" over the 4 channels with 860 SPS.
would give ==> 200++ samples per channel (SPC)
channel 1,2,3 used, channel 4 not connected. 25 % of the performance is not used.

#### Alternative 2:
The continuous mode will "rotate" over the 4 channels with 860 SPS.
would give ==> 200 SPC
suppose you connect channel 1,2,3 as above AND also connect channel 4 to pin 1,2 or 3.
That would allow 2 channel to have 200 SPC and one channel 400 SPC

#### Alternative 3:
The continuous mode will "rotate" over the 4 channels with 860 SPS.
BUT after reading **three** channels you reset it somehow (do not know if and how to do this) and start with channel 1 again.
Assuming this works it would gain some time but it would costs also some time. 
Question is if there is more gain than loss. 
I will dive into this scenario and write some code.  

#### Alternative 4:
If performance need to be higher, consider other ADC  - see my MCP_ADC lib for substantial faster ADC's which uses the SPI bus which is much faster than I2C on average. Those have less resolution, that is the price ...

(to be continued)

RobTillaart, 2023-03-26:
Did some coding and test with 
- an Arduino UNO
- this code (will be added as example, not tested extensively.)
- nothing connected so floating lines.
- has an issue with the index of the channels. not not investigated yet.

```cpp
//
//    FILE: ADS_continuous_3_channel.ino
//  AUTHOR: Rob.Tillaart
// PURPOSE: read multiple analog inputs continuously
//          interrupt driven to catch all conversions.
//     URL: https://github.com/RobTillaart/ADS1X15
//          https://github.com/RobTillaart/ADS1X15/issues/49
//
//  experimental, not tested extensively

// test
// connect multiple potmeters
//
//  RDY ----------------- pin 2 (for IRQ)
//
//  GND ---[   x   ]------ 5V
//             |
//             |
//           ADS(n)
//
//  measure at x  - connect to AIN0..3.
//
//  for the test it is an option to have AIN3 connected to 5V and AIN4 to GND
//  so one can see these as references in the output.
//
//  has an issue with the index of the channels. not not investigated yet.

#include "ADS1X15.h"

// choose your sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);
ADS1115 ADS(0x48);

volatile bool RDY = false;
uint8_t channel = 0;
int16_t val[4] = { 0, 0, 0, 0 };

int SPS = 0;
uint32_t lastTime = 0;

void setup()
{
  Serial.begin(230400);       //  <<<<<<<<<  fast!
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);

  Wire.setClock(800000);

  ADS.begin();
  ADS.setGain(0);        // 6.144 volt
  ADS.setDataRate(7);

  // SET ALERT RDY PIN
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorThresholdLow(0x0000);
  ADS.setComparatorQueConvert(0);

  // SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);

  ADS.setMode(0);            //  continuous mode
  channel = 0;
  ADS.requestADC(channel);   //  start at 0
}

void loop()
{
  //  if conversion ready
  //  request a new one and print the last one.
  if (RDY)
  {
    SPS++;
    val[channel] = ADS.getValue();
    // request next channel asap
    channel++;
    if (channel >= 3) channel = 0;
    ADS.requestADC(channel);
    RDY = false;

    //  to see it works
    if (SPS % 200 == 0)
    {
      for (int i = 0; i < 4; i++)
      {
        Serial.print('\t');
        Serial.print(val[i]);
      }
      Serial.println();
    }
  }

  //  print the SPS
  if (millis() - lastTime >= 1000)
  {
    lastTime = millis();
    Serial.print("SPS: ");
    Serial.println(SPS);
    SPS = 0;
  }
}

//  interrupt service routine
//  kept as minimal as possible
void adsReady()
{
  RDY = true;
}

//  -- END OF FILE --

```

The output was pretty stable at ~831 SPS for 3 channels = ~275 SPC

```
SPS: 831
	3177	3160	3159	0
	3166	3161	3189	0
	3170	3171	3162	0
	3179	3162	3169	0
SPS: 832
	3166	3174	3163	0
	3202	3161	3191	0
	3179	3185	3160	0
	3181	3185	3170	0
SPS: 831
	3166	3185	3175	0
	3190	3155	3157	0
	3180	3178	3189	0
	3189	3202	3171	0
SPS: 831
	3178	3172	3154	0
	3179	3175	3159	0
	3189	3191	3176	0
	3192	3177	3184	0
SPS: 832
```

If I write every line the SPS drops into the 500 something range.

RobTillaart, 2023-03-26:
Added the example above in the develop branch to be merged soon.
(no new release (yet))

RobTillaart, 2023-03-26:
https://github.com/RobTillaart/ADS1X15/pull/50 - PR created ==> green

MopheusDG, 2023-03-26:
Thanks a lot for all the info Rob, I really appreciate it. Regarding to your question:

Q: what board do you use? We are using an Arduino Leonardo Micro. I think these boards have a 400Mhz limit on the I2C, but not sure.

Will give your example a try to see if it runs on my Leonardo micro.

MopheusDG, 2023-03-26:
By the way, on your code I see this 2 lines repeated during the setup, not sure if it's on purpose or not:

```
  pinMode(2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);
```

MopheusDG, 2023-03-26:
And the problem I have with this one is that I don't get any values at all. Probably because of the digitalPinToInterrupt pin, not sure what pin can/shoud I use on Leonardo for this and how to connected in the circuit.

RobTillaart, 2023-03-26:
Probably my mistake, had too many issues under attention today.
(to be continued tomorrow)

MopheusDG, 2023-03-26:
Additional testing:

If I leave 2, 3 or 4 channels in all cases the result is the same (it doesn't matter how many, as soon as I add the channel switching):

```
SPS: 152
	1220	24048	1156	1137
SPS: 153
	1171	24026	1160	1135
SPS: 152
	1164	24038	1105	1146
```

As soon as I remove channel switch, just commenting this 2 lines:

```
    // channel++;
    // if (channel >= 4) channel = 0;
```

```
SPS: 793
	1186	0	0
	1196	0	0
	1226	0	0
	1196	0	0
	1237	0	0
SPS: 792
	1223	0	0
	1198	0	0
	1237	0	0
	1196	0	0
	1259	0	0
SPS: 793
	1227	0	0
	1199	0	0
	1239	0	0
	1194	0	0
	1254	0	0
```

**My conclusion:** It seems like channel switch is returning the SPS values to "default" or, my ADS1115 is wicked :)

RobTillaart, 2023-03-28:
Weird, and you are using the **ADS_continuous_3_channel.ino** I posted above?

> By the way, on your code I see this 2 lines repeated during the setup, not sure if it's on purpose or not:
> 
> ```
>   pinMode(2, INPUT_PULLUP);
>   attachInterrupt(digitalPinToInterrupt(2), adsReady, RISING);
> ```

Changed it in the example in the develop branch, thanks.

MopheusDG, 2023-03-29:
Hi @RobTillaart, yes, I made a couple of changes but nothing significant if I'm not wrong. Will add the code here tomorrow.

By the way, on /examples/ADS_continuous_4_channel/ADS_continuous_4_channel.ino you have another mistype if I'm not wrong, just in the comments., line 50: ADS.setDataRate(7);    // slow

7 is not slow, is faster.

Just my 2 cents :)

Will come back tomorrow with more testing and my code.

Thanks.

RobTillaart, 2023-03-29:
In case of a mismatch between code and comment, code always win. 😁

RobTillaart, 2023-03-29:
Do you have pull-up's on the ALERT pin? It might be that you miss interrupts?

[image]

[image]

MopheusDG, 2023-03-29:
Hi @RobTillaart , yes , I've tested with and without pullup, still stuck on 152 max speed when switching channels, no matter if it's 2, 3 or 4 channels. And tested single shot and continuous mode. No luck unfortunatelly.

RobTillaart, 2023-03-30:
That sounds not good, it mire and more looks like a faulty adc (batch). Maybe contact the supplier?
From the library point of view i see no new ways to make it work.(yet).

MopheusDG, 2023-03-30:
Hi @RobTillaart , yeah, I'm gonna try getting a new ADS from some other supplier cause neither do I, I'm pretty sure it's not the library too, so thanks for your help anyway ! I really appreciate your time and dedication on helping me.

RobTillaart, 2023-03-31:
You're welcome.
Lets keep the issue open until new batch ADS arrives and you tested with them.

RobTillaart, 2023-04-14:
@MopheusDG 
Any progress?

MopheusDG, 2023-04-14:
Hi @RobTillaart unfortunately the new batch has the same issue, but I think is the same kind of board. Can you point us to the one your are using for testing (brand, where you buy it, anything that can help identify differences) so I can check it ?

Thanks ! I think you can close the issue if you want too, there's nothing from the library u can do so far so....

RobTillaart, 2023-04-14:
Buy mine at a local shop in Eindhoven / Netherlands.
- https://www.tinytronics.nl/shop/nl/sensoren/stroom-spanning/ads1115-16-bit-adc-i2c-module-4-kanaals  $6 (approx)
Given the price these are probably batches from China.

Adafruit makes their own boards with QWIIC connectors
- https://www.adafruit.com/product/1085  $15.00
Not as cheap as China but they sell quality boards. They have resellers all over the world.

MopheusDG, 2023-04-14:
Thanks, will check those up to see if can stop something different.

RobTillaart, 2023-04-14:
> anything that can help identify differences

I checked the datasheet if there is a register with type info but there is no such register.
The config register has a default value but that only works if it is not changed (e.g. at boot time) and no guarantee.

- https://forum.arduino.cc/t/ads1115-problems-or-fake/658746  (not read it yet)
-

MopheusDG, 2023-04-25:
Hi, @RobTillaart my actual work has me very busy and couldn't do any more testing or checking on this. Sorry. Will come back soon but need to get some work done (the paying one) :)

RobTillaart, 2023-04-25:
I understand, have that kind of work too

RobTillaart, 2023-05-19:
Did some experiment today

UNO 
IDE 1.8.19

```cpp
//
//    FILE: ADS_continuous.ino
//  AUTHOR: Rob.Tillaart
// PURPOSE: read analog input
//     URL: https://github.com/RobTillaart/ADS1X15

// test
// connect 1 potmeter
//
// GND ---[   x   ]------ 5V
//            |
//
// measure at x (connect to AIN0).

#include "ADS1X15.h"

// choose your sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);
ADS1115 ADS(0x48);

uint16_t count = 0;
uint16_t value = 0;
uint16_t prev  = 0;
uint32_t lastTime = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  ADS.begin();
  ADS.setGain(0);      // 6.144 volt
  ADS.setDataRate(7);  // 0 = slow   4 = medium   7 = fast
  ADS.setMode(0);      // continuous mode
  ADS.readADC(0);      // first read to trigger
}

void loop()
{
  value = ADS.getValue();
  if (value != prev)
  {
    count++;
    prev = value;
    Serial.print(count);
    Serial.print("\t");
    Serial.println(value);
  }
  if (millis() - 1000 > lastTime)
  {
    lastTime = millis();
    count = 0;
  }
}

// -- END OF FILE --
```

output snippet
```
19:33:08.333 -> 823	3138
19:33:08.333 -> 824	3190
19:33:08.333 -> 825	3137
19:33:08.333 -> 826	3138
19:33:08.333 -> 827	3201
19:33:08.380 -> 1	3177
19:33:08.380 -> 2	3172
19:33:08.380 -> 3	3131
19:33:08.380 -> 4	3139
19:33:08.380 -> 5	3158
```

So I came to at least 827 reads per second -  some may be new and equal so it could have been more

RobTillaart, 2023-05-19:
One thing to note is that **Serial.print()** calls also take serious time.
You might see improvements if you 
- minimize the amount of data to print and
- maximize the baudrate of Serial e.g. 500000 iso 115200

MopheusDG, 2023-05-19:
Thanks @RobTillaart will give it a try this weekend to see if there's any improvement !

MopheusDG, 2023-05-21:
Hi @RobTillaart tested your version using 1 potentiometer and I get the same results (861):

```
20:28:39.795 -> 857	908
20:28:39.795 -> 858	915
20:28:39.795 -> 859	924
20:28:39.795 -> 860	925
20:28:39.795 -> 861	925
20:28:39.795 -> 1	941
```

The problem is when adding another potentiometer to the mix. Cause I need to move de readADC(x) to the loop and that's when the SPS goes down.
Can you add to your code and board a second potentiometer and see what results you get ? If do that, I only get like 55 SPS.... but maybe I'm doing something wrong on the code.

Thanks !

RobTillaart, 2023-05-22:
```cpp
//
//    FILE: ADS_continuous.ino
//  AUTHOR: Rob.Tillaart
// PURPOSE: read analog input
//     URL: https://github.com/RobTillaart/ADS1X15

// test
// connect 1 potmeter
//
// GND ---[   x   ]------ 5V
//            |
//
// measure at x (connect to AIN0).

#include "ADS1X15.h"

// choose your sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);
ADS1115 ADS(0x48);

uint16_t count = 0;
uint16_t value = 0;
uint16_t prev  = 0;
uint32_t lastTime = 0;
uint32_t lastSample = 0;

uint8_t  channel = 0;

int  val[4] = {0, 0, 0, 0};

void setup()
{
  Serial.begin(500000);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  ADS.begin();
  ADS.setGain(0);      // 6.144 volt
  ADS.setDataRate(7);  // 0 = slow   4 = medium   7 = fast
  ADS.setMode(0);      // continuous mode
  ADS.readADC(channel);      // first read to trigger
  
  lastTime = micros();
}

void loop()
{
  uint32_t now = micros();
  if (now - lastSample >= 1160)  // almost exact 860 SPS
  {
    lastSample = now;

    val[channel] = ADS.getValue();
    channel++;
    if (channel == 2) channel = 0;
    ADS.requestADC(channel);

    count++;
    if (count % 5 == 0)
    {
      Serial.print(count);
      Serial.print('\t');
      Serial.print(val[0]);
      Serial.print('\t');
      Serial.print(val[1]);
      Serial.print('\t');
      Serial.print(val[2]);
      Serial.print('\t');
      Serial.print(val[3]);
      Serial.println();
    }
  }
  if (now - 1000000 >= lastTime)
  {
    lastTime = now;
    count = 0;
  }
}

// -- END OF FILE --
```

generates still a lot of samples

```
735	3134	3154	0	0
740	3129	3110	0	0
745	3130	3118	0	0
750	3152	3143	0	0
755	3141	3151	0	0
760	3147	3161	0	0
765	3134	3148	0	0
770	3143	3120	0	0
775	3185	3156	0	0
780	3171	3172	0	0
5	3161	3162	0	0
10	3137	3139	0	0
15	3163	3142	0	0
20	3130	3146	0	0
```

To investigate if each channel has right data. 
mmm
lets connect channel 0 to 5V

RobTillaart, 2023-05-22:
Polling mode

```cpp
//
//    FILE: ADS_continuous.ino
//  AUTHOR: Rob.Tillaart
// PURPOSE: read analog input
//     URL: https://github.com/RobTillaart/ADS1X15

// test
// connect 1 potmeter
//
// GND ---[   x   ]------ 5V
//            |
//
// measure at x (connect to AIN0).

#include "ADS1X15.h"

// choose your sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);
ADS1115 ADS(0x48);

uint16_t count = 0;
uint16_t value = 0;
uint16_t prev  = 0;
uint32_t lastTime = 0;
uint32_t lastSample = 0;

uint8_t  channel = 0;

int  val[4] = {0, 0, 0, 0};

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  ADS.begin();
  ADS.setGain(0);      // 6.144 volt
  ADS.setDataRate(7);  // 0 = slow   4 = medium   7 = fast
  ADS.setMode(1);      // continuous mode
  ADS.requestADC(channel);      // first read to trigger

  lastTime = micros();
}

void loop()
{
  uint32_t now = micros();
  if (now - lastSample >= 1160)
  {
    lastSample = now;

    val[channel] = ADS.getValue();
    channel++;
    if (channel == 2) channel = 0;
    ADS.requestADC(channel);

    count++;
    Serial.print(count);
    Serial.print('\t');
    Serial.print(val[0]);
    Serial.print('\t');
    Serial.print(val[1]);
    Serial.print('\t');
    Serial.print(val[2]);
    Serial.print('\t');
    Serial.print(val[3]);
    Serial.println();
  }
  if (now - 1000000 >= lastTime)
  {
    lastTime = now;
    count = 0;
  }
}

// -- END OF FILE --
```

results seem OK, (not tested for long but it look good)

```
550	3413	25576	0	0
551	3413	25576	0	0
552	3394	25576	0	0
553	3394	25575	0	0
554	3399	25575	0	0
555	3399	25578	0	0
556	3430	25578	0	0
557	3430	25580	0	0
558	3418	25580	0	0
559	3418	25577	0	0
560	3404	25577	0	0
561	3404	25578	0	0
562	3393	25578	0	0
1	3393	25576	0	0
2	3397	25576	0	0
3	3397	25575	0	0
4	3402	25575	0	0
5	3402	25572	0	0
6	3430	25572	0	0
7	3430	25575	0	0
8	3434	25575	0	0
9	3434	25569	0	0
10	3395	25569	0	0
```

MopheusDG, 2023-05-22:
Hi @RobTillaart will try using your same code later to see what I get. Thanks.

subhashisbhowmik, 2023-06-13:
Hi, here from #55
@RobTillaart I tried your example (I'm using ESP32 so had to modify your code very slightly) but I'm also not getting anything beyond ~152 SPS.

It looks like a hardware issue or a fake chip?

RobTillaart, 2023-06-13:
My tests were mostly done with UNO's which is a much slower processor. 

Thinking out loud 
If the ESP32 is busy with other tasks e.g. wifi, it might just not have enough time slices for the ADC?
Still as it has two cores it should easily handle it.

No clue, so counterfeit chips could be a cause.

subhashisbhowmik, 2023-06-13:
I don't have anything like wifi or bt enabled. And I made sure to clock just after I set the channel and just after receiving the interrupt, so that should not be a problem. I have ordered new ICs from digikey, but I'll have to solder them and add the pullups manually, as I was using a dev board. Should be able to know if the issue was due to counterfeit ICs in a couple of weeks.

RobTillaart, 2023-06-13:
> Should be able to know if the issue was due to counterfeit ICs in a couple of weeks.

Is it for a commercial project?

subhashisbhowmik, 2023-06-23:
Today I finally received another ADS1115 from a different supplier, I had my oscilloscope hooked up to the ALRT/Ready pin and it was showing 152Hz initially. The moment I connected the new ADS it went to ~860 Hz. It was definitely a faulty or a counterfeit chip!

One thing I don't understand is the markings on the chip. The first one says 8A BOGI, whereas the new working ADS says 78 BOGI. Do you know what this supposedly Hexadecimal number means?

subhashisbhowmik, 2023-06-23:
> Is it for a commercial project?

Just prototyping a device for now, not sure if it'll be commercial or not lol.

RobTillaart, 2023-06-23:
> One thing I don't understand is the markings on the chip. The first one says 8A BOGI, whereas the new working ADS says 78 BOGI. Do you know what this supposedly Hexadecimal number means?

BOGI is part of the packaging

subhashisbhowmik, 2023-06-23:
I am in India, I doubt this will be useful to you. Also FYI, I was using breakout boards, I had also ordered a couple of chips from Digikey, but my soldering skills failed me XD

Faulty/Fake one: [robu.in](https://robu.in/product/i2c-ads1115-16-bit-adc-4-channel-module-with-programmable-gain-amplifier/)
Good one: [robocraze.com](https://robocraze.com/products/16-bit-i2c-4-channel-ads1115-module)

RobTillaart, 2023-06-23:
Thanks for the information.
You are right, small chance I will order in India, however there are many Arduino users in India that might find it useful.

RobTillaart, 2023-06-24:
> One thing I don't understand is the markings on the chip. 
> The first one says 8A BOGI, whereas the new working ADS says 78 BOGI.

@MopheusDG

Do you have any code on your ADS1115?

RobTillaart, 2023-06-24:
Checked mine, they all have 0B BOGI  or 08 BOGI, hard to read. 
These are genuine and working as they should.

subhashisbhowmik, 2023-06-24:
Seems like these are internal code for manufacturing reference, but if they correlate to batch ID or something like that, there might be an issue with the current 8As

MopheusDG, 2023-06-25:
> > One thing I don't understand is the markings on the chip.
> > The first one says 8A BOGI, whereas the new working ADS says 78 BOGI.
> 
> @MopheusDG
> 
> Do you have any code on your ADS1115?

I've uploaded some sshots of mine. Sorry for the delay, my dad had some health issues and I've stopped working on this for a while.

Hope this helps.

[image]
[image]

RobTillaart, 2023-06-25:
If I read it correctly it states 88 BOGI.
The last is a logistic code but the 88 is unknown so far.

MopheusDG, 2023-06-25:
Correct, beter picture here.

[image]

RobTillaart, 2023-06-26:
Thanks, 
'88' could be a batch numbers, although 2 digits would only allow 256 batches which is a very low number.
Other option is that it indicates a production line? then 256 would be reasonable
Final guess is that it indicates production date   88 could mean August 2018 and e.g 7A => October 2017

But, just dont know

googled ADS1115  images on the internet and saw a lot of unsharp photos, a few (barely) readable 7A, 88 8A and a 95(?)
Still no clue about the semantics.

RobTillaart, 2023-08-12:
As this issue has been silent for more than a month I close it.
Feel free to reopen if there is a need / new information to add.

(Just try to reduce the number of open issues in my repositories)

closed 2023-08-12 by RobTillaart