RobTillaart/ADS1X15 #51: Error during compiling

jlcnuke, 2023-04-16:
I'm receiving the following error when compiling a program which uses the ADS1X15.h file:
Compilation error: call to 'bool Adafruit_ADS1X15::begin(uint8_t, TwoWire*)' uses the default argument for parameter 1, which is not yet defined

Specifically it is coming in twice as follows:
In file included from G:\My Drive\My Documents\3D Print Files\Analyzer\files\o2_analyzer\o2_analyzer.ino:25:0:
C:\Users\bridg\OneDrive\Documents\Arduino\libraries\Adafruit_ADS1015/Adafruit_ADS1015.h:157:31: error: expected primary-expression at end of input
   bool begin(uint8_t i2c_addr = ADS1X15_ADDRESS, TwoWire *wire = &Wire);
                               ^
G:\My Drive\My Documents\3D Print Files\Analyzer\files\o2_analyzer\o2_analyzer.ino: In function 'void setup()':
G:\My Drive\My Documents\3D Print Files\Analyzer\files\o2_analyzer\o2_analyzer.ino:98:13: error: call to 'bool Adafruit_ADS1X15::begin(uint8_t, TwoWire*)' uses the default argument for parameter 1, which is not yet defined
   ads.begin(); // ads1115 start
             ^

exit status 1

Can anyone explain how I can resolve this error (I'm very new, sorry if this is an easy fix I just don't know how to do yet)?

Thank you.

jlcnuke, 2023-04-16:
[o2_analyzer.txt](https://github.com/RobTillaart/ADS1X15/files/11244853/o2_analyzer.txt)

RobTillaart, 2023-04-17:
Thanks for opening this issue. I will come back to it later today.

RobTillaart, 2023-04-17:
> Compilation error: call to 'bool Adafruit_ADS1X15::begin(uint8_t, TwoWire*)' uses the default argument for parameter 1, which is not yet defined

Apparently the compiler is using the**Adafruit** library and not mine.

Assuming you are using the Arduino IDE, you must got to the menu FILE->Preferences
You must set the verbose option for compilation and uploading.

[image]

In the output of the compilation you can see which library is used.

To force the use of a specific library move the other (conflicting) libraries "out of sight" of the compiler.
So move it from the libraries folder to  "other_libraries" folder.

RobTillaart, 2023-04-17:
If it is your intend to use the **Adafruit** library, you should open an issue in that repo (or on the Arduino Forum), as then it is not a problem of this library. 

Note: Both AdaFruit and I have a library for this ADS1115 and although functional similar they do not share a compatible interface.

RobTillaart, 2023-04-24:
@jlcnuke 
As there is no further information I assume the issue does not relate to this library and will be closed.

closed 2023-04-24 by RobTillaart