RobTillaart/ADS1X15 #56: Differential continuos mode

tomelgato, 2023-08-16:
Sorry for writing it as an issue ... i did see your example for differential cont mode reading 0-1 & 0-2. And also your discussion about this in issue 29.

Im still about unsure the following fact:

If i read just one pair in continuous mode, do i still need the interrupt?

Cause this doesnt work in my case:

ADS.setMode(CONT);
ADS.requestADC_Differential_0_1();

and in loop

ADS.getValue();

I understand that this could cause reading the same value multiple times without the Int, but does your library work without the int?

and 2nd ... for the interrupt its necessary to connect the ADS int pin to one of the arduino inputs, right?

RobTillaart, 2023-08-16:
No need to apologize, questions are also issues.

To have the maximum number of samples, you need the interrupt. And you are free to sample less frequent.

The continuous mode takes care that the samples are as fresh as they can be, and you choose which to process.

RobTillaart, 2023-08-16:
> If i read just one pair in continuous mode, do i still need the interrupt?

No not necessarily, only if you need / want maximum throughput without polling same sample multiple times.

RobTillaart, 2023-08-16:
> If i read just one pair in continuous mode, do i still need the interrupt?

As discussed in #29 when the device is in continuous mode you cannot use the **isReady()** or **isBusy()** call to check the end of a conversion. 
The device restarts the next conversion immediately after the previous conversion is done, so the device is always "busy".
Only after conversion the interrupt line is triggered to inform the processor, but as said before you do not need to use it.

RobTillaart, 2023-08-16:
> does your library work without the int? 

Yes, 

>In my case i need the sample at a given time, so it doesnt matter if it is the last or a new one.

Just read the value when you need to. 
Don't know your performance requirements, but do you need continuous mode?

tomelgato, 2023-08-16:
Thanks for your extensive answer :)

Right now i can do with single mode, but i would like to experiment with continous. The setup is like that:
Arduino Due, running steppers with accelstepper and doing little bit of background work. The accelstepper library is quite performance hungry, so i try to get the performance footprint low. 
During movement i need to get a sample every X steps. My idea was to just let the ADC run in cont mode and take the sample which is available when i reach the step.
This works but when i go into microstepping my performance window becomes smaller ;)

tomelgato, 2023-08-16:
So the code i posted on top should work, no? I had it implemented but it didnt work as expected during the last machine update, so i switched back to single without further digging. Maybe it was my fault somewhere ;)

tomelgato, 2023-08-16:
Something else ... did you ever get the feeling that after switching mode/channel the first reading(s) are a little bit off with the 1015?

RobTillaart, 2023-08-17:
> Something else ... did you ever get the feeling that after switching mode/channel the first reading(s) are a little bit off with the 1015?

The 1115 in continuous mode seems to always finish the running conversion, so after switching mode or channel there is a big chance you have one sample that is effectively the last with the previous settings.

The clean way to do this is:
1. stop continuous mode, 
2. wait for running conversion to be ready, 
3. change settings, 
4. restart (continuous mode) with new settings.

That takes extra time, and is as robust as you can get.

Maybe I must add some notes in the Readme.md about this.

tomelgato, 2023-08-17:
Thanks a lot, will try!

closed 2023-08-17 by tomelgato