RobTillaart/ADS1X15 #57: Cannot change address

yihan-liu, 2023-08-16:
Hi, I am developing a customized PCB that uses ESP32S3 SoC and communicate with an ADS1115 via I2C on PlatformIO. I happened to have the address of my ADS1115 set to 0x49 by connecting the ADDR pin to VDD. Therefore, I used the following code to define my ADS1115 object:

`ADS1115 ads(0x49);`

I then have the following code to check if the connection is success in setup():

```
if (!ads.isConnected())
{
	ESP_LOGE(TAG, "Failed to initialize ADS1115.");
	while (1);
}
ESP_LOGV(TAG, "ADS1115 initialized.");

```
However, this doesn't seem to work since I am receiving the following message from my serial port:

`[  1206][E][main.cpp:109] sensorInit(): [MY_APP] Failed to initialize ADS1115.`

In contrast, if I use a normal ESP32S3 devkit and a commercial-available ADS1115 module that pulls down ADDR, I can modify the code to either one of the following and make it work:

```
ADS1115 ads(0x48); // First approach
ADS1115 ads; // Second approach
```

The example of the successful output:

`[  1215][V][main.cpp:112] sensorInit(): [MY_APP] ADS1115 initialized.`

I also noticed the following comment in ADS1X15.h

```
//  allow compile time default address
//  address in { 0x48, 0x49, 0x4A, 0x4B }, no test...
#ifndef ADS1015_ADDRESS
#define ADS1015_ADDRESS                   0x48
#endif
```

Does that mean that it is expected to see this behavior with a non-default address?

Thank you very much!

RobTillaart, 2023-08-16:
Thanks for the issue,  will come back too it asap. 
Will try to find some time tomorrow.

RobTillaart, 2023-08-16:
Q: if you connect ADDR to Vdd, and you run an i2c scanner, does it report the address correctly?

yihan-liu, 2023-08-16:
> Q: if you connect ADDR to Vdd, and you run an i2c scanner, does it report the address correctly?

I just tried and it wouldn't detect the address

RobTillaart, 2023-08-17:
> > Q: if you connect ADDR to Vdd, and you run an i2c scanner, does it report the address correctly?
> 
> I just tried and it wouldn't detect the address

Q1: Does it not detect any address?

Q2: Please measure the voltage level of the ADDR pin.
If it is fluctuating it is definitely not connected correctly.

Q3: Please measure the resistance Ω between ADDR pin and VDD (not connected to power).

yihan-liu, 2023-08-17:
> Q1: Does it not detect any address?
>
> Q2: Please measure the voltage level of the ADDR pin. If it is fluctuating it is definitely not connected correctly.
>
> Q3: Please measure the resistance Ω between [...]

A1: No, it's not detecting any address.

A2: It's at a solid 3.29V

A3: The resistance is 234 kΩ.

I just finished these measurement and I am going to try debugging the I2C interface again by running I2C scanner.

RobTillaart, 2023-08-17:
The resistance of 234 is oo high imho.
If you have a direct connection between Vdd and ADDR it should be less than 1 ohm.

If i2c scanner does not see any address the build in initialization of the device failed somehow. At start the device scans which pin is connected to ADDR to set the address (my assumption)

yihan-liu, 2023-08-17:
I do have a pull-up resistor at 10K at ADDR, so that is why the resistance at no power is significant. But you are right that I probably have a faulty PCB.

RobTillaart, 2023-08-17:
You need to connect the vdd direct to addr, so without pull-ups. You might replace the 10K on the pcb with a piece of wire?

yihan-liu, 2023-08-18:
> You need to connect the vdd direct to addr, so without pull-ups. You might replace the 10K on the pcb with a piece of wire?

The I2C scanner could not find the chip no matter how I change the connection (even switching to ADDR being LOW). After a painstakingly amount of test around the ADS1115 I have, I found that I have a fried chip. Switching to a new chip made the I2C scanner find the device at address 0x49. Thanks for the help!

RobTillaart, 2023-08-18:
That was an option I considered however I understood that 0x48 did work? So maybe only the address logic is gone?

Anyway, if your problem is solved you may close the issue.

yihan-liu, 2023-08-18:
> That was an option I considered however I understood that 0x48 did work? So maybe only the address logic is gone?
> 
> Anyway, if your problem is solved you may close the issue.

Yes, on the commercial module the ADDR is pulled to GND. That is why 0x48 works. Anyway, I think at least the code is tested to be good when using address 0x49. Thank you so much!

closed 2023-08-18 by yihan-liu