RobTillaart/ADS1X15 #58: ADS1115 value from one channel is being copied to the other

sghirsst, 2023-08-30:
Hi, I'm developing a project using the ADS1115. I'm facing an issue with copying values between two channels; it seems that the value from one channel is being copied to the other. I'm using a sample rate of 860 fps. Below is the acquireData function for reference:

````cpp
void acquireData(data_t* data) {
   read_data = true; 
   while(read_data){ //wait until the data is read and convReady is true which means that the value is ready.
      if(convReady) //convReady is connected to alertReady PIN
      {
        data->adc[0]  = ADS.getValue();
        ADS.requestADC(1);
        convReady = false;
        read_data = false;
      }
  }

  read_data = true;
  while(read_data){
      if(convReady)
      {
         data->adc[1] = ADS.getValue();
         ADS.requestADC(0);
        convReady = false;
        read_data = false;
      }
  }

}
````
Please note that the first time the program is executed I'm calling ADS.requestADC(0), do you have any suggestions or ideas about this issue.
Thank you in advance.

RobTillaart, 2023-08-30:
Hi, 
Thanks for the question, as I am rather busy on other libraries/ issues it might take a few days. I will try to find some time today and come first suggestions.

Do you run in continuous mode?

sghirsst, 2023-08-30:
All right, thank you, yes I'm running in continuous mode:

````cpp
void userSetup() {
  ADS.begin(21,22); // SDA, SCL
  ADS.setGain(1);
  ADS.setDataRate(7);  // DATA RATE
  ADS.setMode(0);
  // set the thresholds to Trigger RDY pin
  ADS.setComparatorThresholdLow(0x0000);
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorQueConvert(0);             // enable RDY pin !!
  pinMode(4, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(4), convReadyAlert, RISING);
  ADS.requestADC(0);
}
````

sghirsst, 2023-09-01:
Hi, yes convReady its a volatile flag:
`volatile bool convReady = false;`

For a temporary fix we added `delayMicroseconds(500);` is there any alternatives solutions please to this problem ?

sghirsst, 2023-09-04:
Hi, thank you for the alternative solutions 👍

closed 2023-09-04 by sghirsst