RobTillaart/ADS1X15 #61: The use of Wire1 is not allowed for ESP32 or ESP8266

asantiagod, 2023-09-20:
the first line of the function definition "bool ADS1X15::begin(int sda, int scl)" used for ESP32 and ESP8266 microcontrollers forces the use of the "Wire" object, but the constructors allow the selection of the TwoWire object that you want to use.

[image]

RobTillaart, 2023-09-21:
Thanks for the issue. 
Indeed, this needs some investigation.
Will come back to it asap, might take a few days.

RobTillaart, 2023-09-21:
Definitely a bug, the line should not be there. 
Created a develop branch with the line removed. 

If you have time can you verify if the develop branch works as it should?

----

It was also possible to set the pins from Wire1.

```cpp
ADS(0x48, &Wire1);
...
Wire1.begin(sda, scl);
ADS.begin();
```

Maybe I should refactor the whole **begin()** versions as this lib is not responsible for setting the pins?

asantiagod, 2023-09-21:
> Definitely a bug, the line should not be there. Created a develop branch with the line removed.
> 
> If you have time can you verify if the develop branch works as it should?
> 
> It was also possible to set the pins from Wire1.
> 
> ```c++
> ADS(0x48, &Wire1);
> ...
> Wire1.begin(sda, scl);
> ADS.begin();
> ```
> 
> Maybe I should refactor the whole **begin()** versions as this lib is not responsible for setting the pins?

Yes, now it works properly for both cases, that is:
```c++
ADS(0x48, &Wire1);
 ...
Wire1.begin(sda, scl);
ADS.begin();
```
and
```c++
ADS(0x48, &Wire1);
 ...
ADS.begin(sda, scl);
```
Thanks

commit 876ce632 (2023-09-21): fix #61 ESP32 begin()
commit 450fe8a3 (2023-09-22): Develop (#62)

- fix #61 add Wire1 support for ESP32
closed 2023-09-22 by RobTillaart