RobTillaart/ADS1X15 #64: [question] About using ALERT RDY in ADS_continuous_differential.ino example.

ChipCE, 2023-11-20:
Hi, thanks for the great library!
I want to ask about using ALERT RDY in continuous differential mode. 

As the example in [ADS_continuous_differential.ino](https://github.com/RobTillaart/ADS1X15/blob/master/examples/ADS_continuous_differential/ADS_continuous_differential.ino), you called both setComparatorQueConvert(0) and setComparatorThreshold
<pre>  // SET ALERT RDY PIN
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorThresholdLow(0x0000);
  ADS.setComparatorQueConvert(0);</pre>

As I understand, the RDY pin state will sise when the ADC read completed/ready, but what ComparatorThreshold will affect the RDY state here? They will also rise if the signal passed the set thresholds? If I just need the ADC ready signal, just set the <code>ADS.setComparatorQueConvert(0);</code> will be enough?

Sorry for my bad English!
Thanks you!

ChipCE, 2023-11-20:
Hi, I found the answer in the datasheet 
"The conversion-ready function of the ALERT/RDY pin is enabled by setting the Hi_thresh register MSB to 1 and
the Lo_thresh register MSB to 0" 
So the <code>ADS.setComparatorThresholdHigh(0x8000);</code> and <code>setComparatorThresholdLow</code> are just to manually set the register bits and enable the Conversion Ready Pin mode, right? 

I think directly call the <code>_writeRegister</code> is less confusing than<code> setComparatorThreshold</code>. Or should we make a wrapper like enableConverionReadyMode() ? I can submit a PR if you don't mind.

RobTillaart, 2023-11-20:
> Sorry for my bad English!

No need to apologize, it is perfectly understandable! (minor edit to your post)

Been a while since I wrote the examples, I recall that the details around the comparator are not trivial.
So from my head

> but what ComparatorThreshold will affect the RDY state here?
> They will also rise if the signal passed the set thresholds?

The RDY triggers after a conversion, when the value is above the HIGH and resets after LOW threshold.
By setting HIGH threshold < LOW threshold one assures it triggers after every conversion.
Details are in the datasheet, (pictures at page 15?)

> If I just need the ADC ready signal, just set the ADS.setComparatorQueConvert(0); will be enough?

If I recall correctly it depends on the threshold settings, so if they happen to be right (by default or by accident) than it might work. Best check the section of the datasheet.

RobTillaart, 2023-11-20:
you beat me by a second,

> I think directly call the _writeRegister is less confusing than setComparatorThreshold. Or should we make a wrapper like enableConverionReadyMode() ? I can submit a PR if you don't mind.

A wrapper is better than direct register write as these are probably even less clear than current code.
On the other hand the code can be well understood if you read the datasheet, 

An **enable...()** function implies there should also be a **disable...()** function. What should be in that one?
Setting both to zero? From my head the datasheet does not provide clues.

ChipCE, 2023-11-20:
> An **enable...()** function implies there should also be a **disable...()** function. What should be in that one?
> Setting both to zero? From my head the datasheet does not provide clues.
Ah, I didn't think about it before. An enable without disable sounds weird to me too.

I'm thinking about 2 methods.
- 1. Just update the doc and example, insert a note about the set bit things.
- 2. Because we need to setup the bits to use <code>setComparatorQueConvert</code> any way, add the set bit call inside <code>setComparatorQueConvert</code> function is a valid option?

RobTillaart, 2023-11-21:
1. Update doc must be done any way, so yes for that part.
2. Creative solution, need to think about the possible implications. It would introduce a coupling with possibly unforeseen side effects.  Food for thought.

ChipCE, 2023-11-21:
> 1. Update doc must be done any way, so yes for that part.

This is the PR for the doc and example. (I just add a few lines of comments) #65

RobTillaart, 2023-11-21:
PR is appreciated and merged into master. 
A new release is planned for January so it will be included then.

> Tell me if I can help on anything.

Thanks, no specific questions at the moment for this library.

closed 2023-11-21 by ChipCE