RobTillaart/ADS1X15 #68: getMaxVoltage() does not handle ADS1113

Mark-81, 2024-03-01:
Perhaps related to [this](https://github.com/RobTillaart/ADS1X15/issues/2) issue.
ADS1113 has no gain control and the maximum voltage is fixed to ±2.048 V.

Usually, to convert the ADC readings to a voltage you would write:

    float voltage = _ads1113.toVoltage(_ads1113.readADC(0));

but `toVoltage()` calls `getMaxVoltage()` and this does not handle the ADC1113:

```
float ADS1X15::getMaxVoltage()
{
  switch (_gain)
  {
    case ADS1X15_PGA_6_144V: return 6.144;
    case ADS1X15_PGA_4_096V: return 4.096;
    case ADS1X15_PGA_2_048V: return 2.048;
    case ADS1X15_PGA_1_024V: return 1.024;
    case ADS1X15_PGA_0_512V: return 0.512;
    case ADS1X15_PGA_0_256V: return 0.256;
  }
  _err = ADS1X15_INVALID_VOLTAGE;
  return _err;
}
```

because in the inizialization:

```
ADS1113::ADS1113(uint8_t address, TwoWire *wire)
{
  _address = address;
  _wire = wire;
  _config = ADS_CONF_NOCOMP | ADS_CONF_NOGAIN | ADS_CONF_RES_16 | ADS_CONF_CHAN_1;
  _conversionDelay = ADS1115_CONVERSION_DELAY;
  _bitShift = 0;
  _maxPorts = 1;
}
```

the `ADS_CONF_NOGAIN` would prevent any handling of the gain, falling back to a wrong one (`ADS1X15_PGA_6_144V`):

```
void ADS1X15::setGain(uint8_t gain)
{
  if (!(_config & ADS_CONF_GAIN)) gain = 0;
  switch (gain)
  {
    default:  //  catch invalid values and go for the safest gain.
    case 0:  _gain = ADS1X15_PGA_6_144V;  break;
    case 1:  _gain = ADS1X15_PGA_4_096V;  break;
    case 2:  _gain = ADS1X15_PGA_2_048V;  break;
    case 4:  _gain = ADS1X15_PGA_1_024V;  break;
    case 8:  _gain = ADS1X15_PGA_0_512V;  break;
    case 16: _gain = ADS1X15_PGA_0_256V;  break;
  }
}

uint8_t ADS1X15::getGain()
{
  if (!(_config & ADS_CONF_GAIN)) return 0;
  switch (_gain)
  {
    case ADS1X15_PGA_6_144V: return 0;
    case ADS1X15_PGA_4_096V: return 1;
    case ADS1X15_PGA_2_048V: return 2;
    case ADS1X15_PGA_1_024V: return 4;
    case ADS1X15_PGA_0_512V: return 8;
    case ADS1X15_PGA_0_256V: return 16;
  }
  _err = ADS1X15_INVALID_GAIN;
  return _err;
}
```

To ensure compatibility of the library across all the supported devices, I think it should handle the ADS1113 properly and `getMaxVoltage()` should return 2.048, perhaps forcing the "gain" to `ADS1X15_PGA_2_048V`.

RobTillaart, 2024-03-01:
Thanks for the issue, 
Need to dive into the details / data sheet, had a quick look and I think I understand the problem (bug).
Coming days I will be rather busy so I try to free time for this coming week.

RobTillaart, 2024-03-01:
Note 
- ADS1013 idem
- need tracking type ?

RobTillaart, 2024-03-04:
@Mark-81
Gave it some thought and read some datasheet pages and came to the following fix which is minimalistic.
The user cannot set the _gain anymore and the maxVoltage() will now return 2.048 as it should.

```cpp
//  ADS1x13 has no gain so set default.
//  Table 8. Config Register Field Descriptions
void ADS1113::setGain(uint8_t gain)
{
  _gain = ADS1X15_PGA_2_048V;  //  fixed
}
```

Will push the develop branch in a few minutes.

RobTillaart, 2024-03-04:
@Mark-81 
Added some more code to make it robust. Added an example to test it.
pushed the develop branch 

- should add a unit test for this.

RobTillaart, 2024-03-04:
@Mark-81 
Develop branch looks stable now, can you test if it works now for the ADS1113?
Thanks,

Mark-81, 2024-03-04:
@RobTillaart yeah, it seems to work fine now! Thanks for your quick fix!

commit 9dbbb9a4 (2024-03-04): fix #68, gain for ADS1x13
commit 09745296 (2024-03-04): fix #68, gain for ADS1x13 (#69)

- fix #68, gain bugs ADS1x13
- add unit test for ADS1x13
- update GitHub/actions to v4
- removed depreciated **getLastValue()**
- add multiplexer section to readme.md
closed 2024-03-04 by Mark-81