RobTillaart/ADS1X15 #70: interrupt driven reads

holgerlembke, 2024-03-05:
Moin Rob.

Thanks for the nice lib.

Just a note, not a real issue.

I was looking for an example of interrupt driven conversions. Looking into the example file name list, I found nothing. Looking into the main .MD page: again nothing. 

It needed a text search throu the examples.

Not that a big deal, but from a "marketing" standpoint I think interrupt driven conversions (like in ADS_continuous_4_channel) should be displayed very prominently. Isn't it what we all want?

mfg.

RobTillaart, 2024-03-05:
Thanks for the thumps up

Interrupt driven, mmmm
Imho professional users known when they really need it. These are the people that read datasheets.
Often proper time management solves the issue.
That said, if you want to maximize throughput you need to know how interrupts work.

Marketing is a sword with too edges, focus on the interrupt side may be scary too for the non pro.
For me there is little? / no gain in marketing

holgerlembke, 2024-03-05:
Misunderstanding.

My "complain" is simply that it is not soooo easy to see that this feature exists and in what example it is shown.

RobTillaart, 2024-03-05:
I will add a few lines in the readme.md for the next update.

closed 2024-03-05 by holgerlembke

RobTillaart, 2024-03-05:
This is the list of examples that use the RDY pin

```
ADS_1114_two_continuous.ino
ADS_continuous_3_channel.ino
ADS_continuous_4_channel.ino
ADS_continuous_8_channel.ino
ADS_continuous_differential.ino
ADS_high_speed_differential.ino
ADS_read_async_rdy.ino
ADS_read_RDY.ino
```

holgerlembke, 2024-03-05:
hehehehe, I'm an professional users and know the search icon in VSC. :-)

RobTillaart, 2024-03-05:
Additional section, just after the first table

#### Interrupts

Besides polling the ADS1x14 and ADS1x15 support interrupts to maximize throughput 
with minimal latency. For this these device has a ALERT/RDY pin. 

This ALERT/ RDY pin can be used for interrupts or polling, see table below.

|   example                         |  Interrupts  |  notes  |
|:---------------------------------:|:------------:|:-------:|
|  ADS_1114_two_continuous.ino      |      Y       |
|  ADS_continuous_3_channel.ino     |      Y       |
|  ADS_continuous_4_channel.ino     |      Y       |
|  ADS_continuous_8_channel.ino     |      Y       |
|  ADS_continuous_differential.ino  |      Y       |
|  ADS_high_speed_differential.ino  |      Y       |
|  ADS_read_async_rdy.ino           |   polling    |
|  ADS_read_RDY.ino                 |   polling    |