RobTillaart/ADS1X15 #71: Differential input and gain for maximum precision.

ChSt15, 2024-03-14:
Hi, I have a question about the gain and differential. There is a voltage that varies around a bias. I want to measure this voltage in reference to the bias with as much precision as possible. If I use the differential input will the gain setting amplify the result of the differential and therefore improve precision or will it amplify the inputs and then get the differential voltage?

RobTillaart, 2024-03-14:
Good question, never tried. I expect that it will as gain affects single measurements too. Does the datasheet say anything here?

Note that would always be an option to do the differential yourself with two single adc calls with optimum gain per channel.

If you want the differential at the exact same time you might consider using two ADS1x15 devices side by side.

lynx985, 2024-03-23:
> Hi, I have a question about the gain and differential. There is a voltage that varies around a bias. I want to measure this voltage in reference to the bias with as much precision as possible. If I [...]

look at the datasheet on page 14
the input signals get muxed to the PGA (gain) and then converted
if you have an offset on the signals, it will only amplify the difference between the signals
(as long as the offset is within the input voltage aka supply voltage range ofc)

ChSt15, 2024-03-24:
Ok thanks. So I guess this means that I could dynamically increase the gain at lower input diffferentials to improve precision and lower gain to get a larger input range at cost of lower precision.

closed 2024-03-24 by ChSt15

RobTillaart, 2024-03-24:
> Ok thanks. So I guess this means that I could dynamically increase the gain at lower input diffferentials to improve precision and lower gain to get a larger input range at cost of lower precision.

Yes and no.
if your input exceeds what the device can handle you need to add a voltage divider before the inputs.
Or in case of differential probably two equal ones (with high precision resistors)