RobTillaart/ADS1X15 #73: 10-12 Channel Continuous with interrupts.

marcus-4, 2024-04-04:
Hi Rob, all your work on this is much appreciated! By far the most comprehensive implementation for these devices! 

I am trying to sample 10-12 channels (3x ADS1115) as quickly as possible. Using 860sps, 6.144 FSR, 5v supply, Arduino Mega2560 (I think I should be able to use 3 interrupt pins - 2, 3, and 18 - per #14 ). My current setup is based off of the ADS_continuous_8_channel.ino example, I simply copied/pasted to create ADS_3. Currently, the code causes a hang with 3 ADCs, I believe due to arbitration issues on the I2C bus. With the third ADC commented out, it works as expected, because it is pretty much identical to the example. The attached code has 3 channels on ADS_1 and ADS_2, and 4 channels on ADS_3. For this reason, I set the address of ADS_3 to 0x48, in hopes that it would win arbitration to make up for the slightly slower sampling (3 vs 4 channels). I was using 3 channels on the other 2 ADCs in an attempt to take advantage of some advantages discussed in #49 - the hope was to sample 10 channels with maximum SPS.

In terms of solutions, the following comment from #53  seems like it may have potential: "What might be an option is to combine async requesting a conversion and use interrupts to trigger the reading of the result. Polling a flag is faster than requesting status over I2C."

I have also played around with the ADS_async_8_channel.ino example (and 16channel variant), #24 . These examples (of course) work flawlessly for my application- but my initial goal was to reach faster sampling rates by modifying the 8channel_continuous code instead. 

Should I stick with the async implementation? Is this the fastest performance possible with 3x ADS1115? Thank you!
Code:

```

uint32_t starttime1;
uint32_t endtime1;
uint32_t starttime2;
uint32_t endtime2;
uint32_t starttime3;
uint32_t endtime3;
uint32_t times[10] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

int SPS1 = 0;
int SPS2 = 0;
int SPS3 = 0;
uint32_t lastTime1 = 0;
uint32_t lastTime2 = 0;
uint32_t lastTime3 = 0;

#include "ADS1X15.h"

//  adjust addresses if needed
ADS1115 ADS_1(0x49); //VDD 1,2,3
ADS1115 ADS_2(0x4B); //SCL 4,5,6
ADS1115 ADS_3(0x48); //GND 7,8,9,10 - lowest because 4 channels, will win arbitration

//  two interrupt flags
volatile bool RDY_1 = false;
volatile bool RDY_2 = false;
volatile bool RDY_3 = false;

uint8_t channel_1 = 0;  //  channel from device 1
uint8_t channel_2 = 0;  //  channel from device 2
uint8_t channel_3 = 0;  //  channel from device 2

//  array to hold the data.
int16_t val[10] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

void setup() {

  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  Wire.begin();

  //  SETUP FIRST ADS1115
  ADS_1.begin();
  ADS_1.setGain(0);      //  6.144 volt
  ADS_1.setDataRate(7);  //  0 = slow   4 = medium   7 = fast
  ADS_1.setWireClock(400000);

  //  SET ALERT RDY PIN
  ADS_1.setComparatorThresholdHigh(0x8000);
  ADS_1.setComparatorThresholdLow(0x0000);
  ADS_1.setComparatorQueConvert(0);

  //  SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(18, INPUT_PULLUP); //ADC1 whjte wire
  attachInterrupt(digitalPinToInterrupt(18), adsReady_1, RISING);

  ADS_1.setMode(0);             //  continuous mode
  ADS_1.requestADC(channel_1);  //  trigger first read

  //  SETUP SECOND ADS1115
  ADS_2.begin();
  ADS_2.setGain(0);  //  6.144 volt
  ADS_2.setDataRate(7);
  ADS_2.setWireClock(400000);

  //  SET ALERT RDY PIN
  ADS_2.setComparatorThresholdHigh(0x8000);
  ADS_2.setComparatorThresholdLow(0x0000);
  ADS_2.setComparatorQueConvert(0);

  //  SET INTERRUPT HANDLER TO CATCH CONVERSION READY
  pinMode(2, INPUT_PULLUP); //ADC2 yellow wire
  attachInterrupt(digitalPinToInterrupt(2), adsReady_2, RISING);

  ADS_2.setMode(0);             //  continuous mode
  ADS_2.requestADC(channel_2);  //  trigger first read

 //  SETUP THIRD ADS1115
 ADS_3.begin();
 ADS_3.setGain(0);      //  6.144 volt
 ADS_3.setDataRate(7);  //  0 = slow   4 = medium   7 = fast
 ADS_3.setWireClock(400000);

 //  SET ALERT RDY PIN
 ADS_3.setComparatorThresholdHigh(0x8000);
 ADS_3.setComparatorThresholdLow(0x0000);
 ADS_3.setComparatorQueConvert(0);

 //  SET INTERRUPT HANDLER TO CATCH CONVERSION READY
 pinMode(3, INPUT_PULLUP); //ADC3 purple wire
 attachInterrupt(digitalPinToInterrupt(3), adsReady_3, RISING);

 ADS_3.setMode(0);             //  continuous mode
 ADS_3.requestADC(channel_3);  //  trigger first read
}

void loop() {
  handleConversion();

  //mine
  String serial_out = "";
  String time_out = "";

  Serial.println("DEBUG: Entering Loop");
  for (int i = 0; i < 10; i++) {
    serial_out = serial_out + '\t' + val[i];
    time_out = 't' + time_out + '\t' + times[i];
    //Serial.print(val[i]);
    //Serial.print('\t');
    handleConversion();
  }
  //Serial.println();

  //Mine
  Serial.println(serial_out);
  Serial.println(time_out);
  
}

//  catch interrupt and set flag device 1
void adsReady_1() {
  RDY_1 = true;
}

//  catch interrupt and set flag device 1
void adsReady_2() {
  RDY_2 = true;
}

void adsReady_3() {
  RDY_3 = true;
}

//  handle conversions that are ready
bool handleConversion() {
  bool rv = false;
  if (RDY_1) {
    //  save the last value
    SPS1++;
    val[channel_1] = ADS_1.getValue();

    endtime1 = micros();
    times[channel_1] = endtime1 - starttime1;

    //  request next channel
    channel_1++;
    if (channel_1 >= 3) channel_1 = 0;

    starttime1 = micros();

    ADS_1.requestADC(channel_1);
    RDY_1 = false;
    rv = true;

    
    if (SPS1 % 200 == 0) {
      for (int i = 0; i < 3; i++) {
        Serial.print('\t');
        Serial.print(val[i]);
      }
      Serial.println();
    }

    if (millis() - lastTime1 >= 1000) {
      lastTime1 = millis();
      Serial.print("SPS1: ");
      Serial.println(SPS1);
      SPS1 = 0;
    }
    

  }
  if (RDY_2) {
    SPS2++;
    //  save the last value
    val[3 + channel_2] = ADS_2.getValue();

    endtime2 = micros();
    times[3 + channel_2] = endtime2 - starttime2;

    //  request next channel
    channel_2++;
    if (channel_2 >= 3) channel_2 = 0;

    starttime2 = micros();

    ADS_2.requestADC(channel_2);
    RDY_2 = false;
    rv = true;

    
    if (SPS2 % 200 == 0) {
      for (int i = 3; i < 6; i++) {
        Serial.print('\t');
        Serial.print(val[i]);
      }
      Serial.println();
    }
    if (millis() - lastTime2 >= 1000) {
      lastTime2 = millis();
      Serial.print("SPS2: ");
      Serial.println(SPS2);
      SPS2 = 0;
    }
    
  }
 if (RDY_3) {
   SPS3++;
   //  save the last value
   val[6 + channel_3] = ADS_3.getValue();

   endtime3 = micros();
   times[6 + channel_3] = endtime3 - starttime3;

   //  request next channel
   channel_3++;
   if (channel_3 >= 4) channel_2 = 0;

   starttime3 = micros();

   ADS_3.requestADC(channel_3);
   RDY_3 = false;
   rv = true;
   
   if (SPS3 % 200 == 0) {
     for (int i = 6; i < 10; i++) {
       Serial.print('\t');
       Serial.print(val[i]);
     }
     Serial.println();
   }
   if (millis() - lastTime3 >= 1000) {
     lastTime3 = millis();
     Serial.print("SPS3: ");
     Serial.println(SPS3);
     SPS3 = 0;
   }
   
   
 }

  return rv;
}

```

RobTillaart, 2024-04-05:
A first look through the code does not reveal any big improvements.

You could comment the print statements in **loop()** as those are printed under interrupt which may slow down things a bit.
Minimize the other print statements and use Serial.begin(500000); would save some clock cycles.

Furthermore you could skip one call to microseconds(). would maybe add 1 sample per second.
```cpp
  if (RDY_2) {
    SPS2++;
    //  save the last value
    val[3 + channel_2] = ADS_2.getValue();

    endtime2 = micros();
    times[3 + channel_2] = endtime2 - starttime2;

    //  request next channel
    channel_2++;
    if (channel_2 >= 3) channel_2 = 0;

    starttime2 = endtime2;  //  <<<<<<<<<<<<
    ADS_2.requestADC(channel_2);
```

removing the times capture would save some clock cycles too.

you could make ```uint8_t channel_2``` an int as that is the default datatype. 
Don't know if the compiler already optimizes this (gains might be a few micros.)

So you might gain a few cycles but I expect no substantial increase in SPS.

marcus-4, 2024-04-06:
Okay, thanks for taking a look! Besides SPS, My more significant issue is that it does not currently function with 3 ADCs, the sketch hangs after 2-3 readings. I believe the code is missing safety protections, but I'm not exactly sure how to implement them.

I was also hoping you could provide a bit more detail on how your "continuous" examples are different from the "async" examples, I don't think I fully understand.

marcus-4, 2024-04-06:
Still needing help to make async safe continuous interrupts.

RobTillaart, 2024-04-06:
As your question is not solved yet I reopened the issue, so I see it on my open issues list.

lynx985, 2024-04-07:
hi, i did some speed testing in the past, on a adafruit feather adalogger M0 a SAMD21 chip running at 48 MHz
at 400 kHz: 
requesting a sample takes around: 120 us
and reading a sample takes round: 150 us
so together around 270 us 
however on an atmega board running at 16 MHz I guess it adds around 60-80 us for a total of around 340 us

i was able to run the ADS1115 up to 1.2 MHz (short lines, 1.5 k pullups, may vary from device to device)
but I would no recommend to go higher than 600 - 800 kHz (ideally test the limit and reduce by a factor of 1.5 or 2)
on an atmega I think above around 600 - 800 kHz one gets diminishing returns anyways

@600 kHz (on an M0): 
request: 80 us
reading: 100 us
so a total of 180 us which on a atmega might end up at give or take 250 us

RobTillaart, 2024-04-07:
@lynx985 
Thanks for these performance figures. Insightful

If time permits and I can find a free MEGA I will do actual timing measurements this week.

RobTillaart, 2024-04-07:
Could do a simple test on an UNO, which is not a MEGA but same clock speed and same ARCH
Performance is better than I expected, see table below.

|  clock  |  requestADC () |  getValue()  |  sum  |  notes  |
|:-------:|:--------------:|:------------:|:-----:|:--------|
|   100   |     444        |     564      | 1008  |
|   200   |     252        |     324      |  576  |
|   400   |     156        |     196      |  352  |
|   600   |     124        |     156      |  280  |  max advised
|   800   |     112        |     140      |  252  |
|  1000   |      56        | lock / crash |       |

Code snippet
```cpp
  start = micros();
  ADS.requestADC(0);
  stop = micros();
  Serial.println(stop - start);
  delay(100);
  while (ADS.isBusy());
  start = micros();
  val_0 = ADS.getValue();
  stop = micros();
  Serial.println(stop - start);
  delay(100);
```

RobTillaart, 2024-04-22:
@marcus-4 

Have you tried the proposed improvements?

> * minimize what is printed (e.g nly SPS count)
> * increase Serial baudrate to 250000
> * increase I2C clock to 500k or 600k

RobTillaart, 2024-04-26:
As there are no new insights here I close the issue.
Feel free to reopen if needed.

closed 2024-04-26 by RobTillaart