RobTillaart/ADS1X15 #74: possible error in the readme

lynx985, 2024-06-22:
hi

i tried to implement the rdy pin on an ads1115 but it only worked half the time and depended on input voltages on the adc. 
this made me suspicious that it might not work as a ready pin but as a comparator

so i read the datasheet, where it says:
"Set the most-significant bit of the Hi_thresh register to 1 ... "
I believe this is a 16 bit value because it is a 16 bid ADC, so i think this would be 0x8000  ?

in the readme it says:
"If the thresholdHigh is set to 0x0100  and the thresholdLow to 0x0000 the ALERT/RDY pin is triggered when a conversion is ready."

0x0100 did not really work but 0x8000 seems to work so far

i might messed up something else because the ready pin goes high after a conversion insted of low
the datasheet says if comp_pol is 0 it goes low, but it goes high and low if set to 1

kind regards

code for setting up the ads1115
  ADS.begin();
  ADS.setGain(1);        
  ADS.setDataRate(7);    //  0 = slow   4 = medium   7 = fast
  ADS.setMode(1);  //triggered mode
  ADS.setComparatorThresholdLow(0x0000);  // all zero for RDY function
  ADS.setComparatorThresholdHigh(0x8000);  // 0x8000 for RDY function
  ADS.setComparatorQueConvert(0);        //  enable RDY pin after one conversion
  ADS.setComparatorLatch(0);  // no latch
  ADS.setComparatorPolarity(0); // make sure it works as RDY pin (maybe not needed, but datasheet says it goes low if this is 0)

  ADS.requestADC_Differential_1_3(); //do stuff to write commands above

RobTillaart, 2024-06-22:
Thanks for the issue, I will look into it coming week.

RobTillaart, 2024-06-22:
quick check shows you're right, good catch.

_9.3.8 Conversion Ready Pin (ADS1114 and ADS1115 Only)
The ALERT/RDY pin can also be configured as a conversion ready pin. Set the most-significant bit of the
Hi_thresh register to 1 and the most-significant bit of Lo_thresh register to 0 to enable the pin as a conversion
ready pin._

RobTillaart, 2024-06-25:
Created a develop branch to fix this. 
Need to review the code of the library and examples if the register is used incorrectly.

RobTillaart, 2024-06-25:
> I might messed up something else because the ready pin goes high after a conversion insted of low
the datasheet says if comp_pol is 0 it goes low, but it goes high and low if set to 1

I had to reread the datasheet (again) to understand (I hope) the details.

_When configured as a conversion ready pin, ALERT/RDY continues to require a **pullup resistor**. The ADS111x provide
an approximately 8-μs conversion ready pulse on the ALERT/RDY pin at the end of each conversion in
**continuous-conversion** mode, as shown in Figure 29. In **single-shot** mode, the ALERT/RDY pin asserts low at the
end of a conversion if the COMP_POL bit is set to 0._

[image]

#### Continuous Mode

According to the image the conversion is ready when the 8 us pulse goes **LOW**.
So the interrupt must watch for a **FALLING** edge (5V => 0V) to be sure. As it might take some time before 
fetching data from the ADS, you might gain the 8us by using the **RISING** edge.

#### Single Shot Mode

[image]

In the single shot mode the ALERT/RDY pin goes LOW if the COMP_POL bit is set to 0, this is also the default value / behavior.
According to this description in the table 8 (above) the ALERT/RDY pin should go HIGH if the COMP_POL bit is set to 1. 

So this is how I understand the working: (no HW nearby to test)
- If you set COMP_POL to 0, the ALERT/RDY pin is pulled HIGH by the pullup resistor, and will go LOW when conversion is ready. It is unclear to me if it is a short LOW pulse, or that the ALERT/RDY pin stays LOW until next single shot conversion starts.
- if you set COMP_POL to 1, the ALERT/RDY pin seems to be pulled LOW by the ADS, and will give a HIGH LOW pulse when conversion is ready. This gives a similar pulse as in the continuous mode.

RobTillaart, 2024-06-25:
@lynx985 
fixed the readme.md  + checked the examples in develop branch
No functional code change
Will create a PR from develop branch and merge it later this week.

RobTillaart, 2024-06-26:
@lynx985 
The readme.md file is updated, thanks again for your sharp eyes!

----

If the COMP_POL problem still exists, please open a separate issue for it.
I need a hardware setup to verify the problem so I want to handle it as a separate issue.

closed 2024-06-26 by RobTillaart

lynx985, 2024-06-26:
thx for your work

commit 6ba6f8e9 (2024-06-25): Fix #74, update readme.md