RobTillaart/ADS1X15 #76: COMP_POL reversed

lynx985, 2024-06-26:
so did a short test

measured the RDY-pin on an oscilloscope 

in single shot mode RDY remains high/low until the next request is made (not just a pulse) 

but COMP_POL seems to be switched 
COMP_POL (0) goes LOW while convertig (around 1.2 ms) and HIGH when ready for as long as one wishes
COMP_POL(1) goes HIGH while convertig (around 1.2 ms) and LOW when ready

RobTillaart, 2024-06-26:
Thanks for these tests, 
Need to confirm these so it might take (serious) time before I have a HW setup.

RobTillaart, 2024-06-28:
@lynx985 

First results of 4 test runs.
```
  //  test polarity
  //  MODE                COMP_POL    ALERT/RDY PIN
  //  0 = continuous      0 = LOW     LOW with 8 us HIGH pulse
  //  0 = continuous      1 = HIGH    HIGH with 8 us LOW pulse
  //  1 = single          0 = LOW     continuous HIGH
  //  1 = single          1 = HIGH    continuous LOW
```

In single mode I get no pulses (yet).

Test sketch
```cpp
//
//    FILE: ADS_COMP_POL.ino
//  AUTHOR: Rob.Tillaart
// PURPOSE: read analog input
//     URL: https://github.com/RobTillaart/ADS1X15/issues/76

#include "ADS1X15.h"

// choose your sensor
// ADS1013 ADS(0x48);
// ADS1014 ADS(0x48);
// ADS1015 ADS(0x48);
// ADS1113 ADS(0x48);
// ADS1114 ADS(0x48);

ADS1115 ADS(0x48);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  Wire.begin();

  ADS.begin();
  ADS.setGain(0);  //  6.144 volt

  //  test polarity
  //  MODE                COMP_POL    ALERT/RDY PIN
  //  0 = continuous      0 = LOW     8 us HIGH PULSE
  //  0 = continuous      1 = HIGH    8 us LOW pulse
  //  1 = single          0 = LOW     continuous HIGH
  //  1 = single          1 = HIGH    continuous LOW
  ADS.setMode(1);
  ADS.setComparatorPolarity(1);
  ADS.setComparatorThresholdHigh(0x8000);
  ADS.setComparatorThresholdLow(0x1000);
  ADS.setComparatorQueConvert(0);

  Serial.println("Voltage");
}

void loop()
{
  int16_t raw = ADS.readADC(0);
  delay(1000);
  Serial.println(ADS.toVoltage(raw), 3);
  delay(1000);
}

//  -- END OF FILE --
```

RobTillaart, 2024-06-28:
@lynx985 

Fetched a better scope to check for pulses - see 3 screenshots below - resulted in the following observations.

| TEST |  MODE            |  COMP_POL  |  ALERT/RDY PIN                |  Notes  |
|:----:|:-----------------|:-----------|:------------------------------|:--------|
|   1  |  0 = continuous  |  0 = LOW   |  LOW with 8 us HIGH pulse     |  micro second pulse as specified in datasheet
|   2  |  0 = continuous  |  1 = HIGH  |  HIGH with 8 us LOW pulse     |  micro second pulse as specified in datasheet
|   3  |  1 = single      |  0 = LOW   |  HIGH with an 8 ms LOW pulse  |  1000 x longer pulse
|   4  |  1 = single      |  1 = HIGH  |  LOW with an 8 ms HIGH pulse  |  1000 x longer pulse

#### Observations:
- polarity in single mode seems indeed inverted
- length of the pulse in continuous mode and in single mode differs a factor 1000.

----

## Screenshots

### MODE = 0, COMP_POL = 1

[image]

----

### MODE = 1, COMP_POL = 0

[image]

----

### MODE = 1, COMP_POL = 1

[image]

RobTillaart, 2024-06-28:
TODO: add note in the readme.md

RobTillaart, 2024-06-28:
Created a new develop branch with updated readme.md (and some minor edits).

RobTillaart, 2024-06-28:
> so did a short test
>
> measured the RDY-pin on an oscilloscope
>
> in single shot mode RDY remains high/low until the next request is made (not just a pulse)
>
> but COMP_POL seems to be switched COMP_POL [...]

Can you post your test sketch, so I can confirm (understand) your observations as these are slightly different than mine.

What board do you ise?

RobTillaart, 2024-06-29:
Note that my above observations are made with a fixed data rate and synchronuous calls.
Todo: investigate effect data rate
Todo:: investigate async calls.

RobTillaart, 2024-06-29:
### Investigate effect data rate

Same sketch as above + **ADS.setDataRate()**
- MODE = 1 (Single shot modus)
- COMP_POL = 1   
- DATA_RATE = 0..7

|  datarate  |  pulse length  |
|:----------:|:--------------:|
|      0     |     125 ms     |
|      1     |      62 ms     |
|      2     |      32 ms     |
|      3     |      16 ms     |
|      4     |       8 ms     |
|      5     |     4.4 ms     |
|      6     |     2.4 ms     |
|      7     |     1.2 ms     |

Quick estimates from scope.

RobTillaart, 2024-06-29:
### Investigate level before any call 

Same sketch as above minus code in ```loop()```

- MODE = 1 (Single shot modus)
- COMP_POL = 1
- DATA_RATE = 4

After power off and no request the polarity is indeed HIGH.

- MODE = 1 (Single shot modus)
- COMP_POL = 0
- DATA_RATE = 4

After power off and no request the polarity is still HIGH.

mmmm, lets think....

----

aha, settings are not activated as the library does that in the requestADC() call

Added call to ADS.requestADC() at end of setup() 

COMP_POL = 0 => gave HIGH with a 8 ms LOW pulse (as before).
COMP_POL = 1 => gave LOW with a 8 ms HIGH pulse

### Conclusions

- COMP_POL is indeed behaving inverted in single shot mode.
- conversion ready always gives a pulse
- the length of the pulse depends on the data rate in single shot mode.

RobTillaart, 2024-06-29:
@lynx985 

Found the right phrasing for the semantics of COMP_POL:

- if COMP_POL = 0, a FALLING edge indicates conversion ready.
- if COMP_POL = 1, a RISING edge indicates conversion ready. 

This is true for both continuous and single mode.

RobTillaart, 2024-06-29:
@lynx985 
Updated the readme.md with all the detailed insights from above.
Took more time than expected and imho it was worth the effort to understand these details
New version 0.4.4 will be (expect) this weekend.

closed 2024-06-30 by RobTillaart

lynx985, 2024-07-01:
thanks for all your work =)

RobTillaart, 2024-07-02:
@lynx985 
My latest insights are to extend the description as follows:

- if COMP_POL = 0, 
  - a FALLING edge indicates conversion start.
  - a LOW level indicates converting.
  - a RISING edge indicates conversion ready.
  - a HIGH level indicates idle.
 
- if COMP_POL = 1, 
  - a RISING  edge indicates conversion start.
  - a HIGH level indicates converting.
  - a FALLING edge indicates conversion ready.
  - a LOW level indicates idle.
 
This seems to match with all the scope images above and observations including those in your first post.

For the continuous mode it doesn't matter (much) which edge to take as they are only 8 us apart. 
However for the single shot mode my first thoughts as described in the readme seems not right.

commit 6c86d80d (2024-06-30): update readme.md COMP_POLARITY (#77)

- Fix #76, update readme.md Comparator Polarity
- added defines to replace magic numbers (not used in code yet)
- minor edits
commit ebf49f37 (2024-07-04): refactor #78, redo #76 (#79)

- Fix #78, prevent infinite loop. (Thanks to devmirek).
- Fix #76 (again), update readme.md Comparator Polarity.
- add ADS1X15_ERROR_I2C, communication error.
- add minimal section in readme.md about error codes.
- minor edits.