RobTillaart/ADS1X15 #80: setComparatorPolarity and setComparatorLatch are inverting the setting?

HineSite, 2024-08-20:
I could be wrong, but I don't think these are correct?

`
void ADS1X15::setComparatorPolarity(uint8_t pol)
{
  _compPol = pol ? 0 : 1;
}

void ADS1X15::setComparatorLatch(uint8_t latch)
{
  _compLatch = latch ? 0 : 1;
}
`
Based on how they registers are being set later:
`
  if (_compPol)   config |= ADS1X15_COMP_POL_ACTIV_HIGH;      //  bit 3      ALERT active value
  else            config |= ADS1X15_COMP_POL_ACTIV_LOW;
  if (_compLatch) config |= ADS1X15_COMP_LATCH;
  else            config |= ADS1X15_COMP_NON_LATCH;           //  bit 2      ALERT latching
`

Example:
`
  ADS.setComparatorPolarity(1);
  ADS.setComparatorLatch(1);

  ADS.readADC(0);

  Serial.println(
    "Polarity: " + String(ADS.getComparatorPolarity()) + 
    " | Latch: " + String(ADS.getComparatorLatch())
  );
`
Results: Polarity: 0 | Latch: 0

RobTillaart, 2024-08-20:
Thanks for the issue, 
As I am quite busy on some other tasks, it might take several days (weeks) before I can dive into it.
Definitely looks like a bug given your code snippet, so I labeled it as such to give it priority it needs.

RobTillaart, 2024-08-20:
@HineSite 
Can you run this example as I have to test setup near me.
Please post the output.

It checks all the "configuration flags".

```cpp
//
//    FILE: ADS_test_config.ino
//  AUTHOR: Rob.Tillaart
// PURPOSE: test config flags
//     URL: https://github.com/RobTillaart/ADS1X15
//          triggered by issue 80

#include "ADS1X15.h"

ADS1115 ADS(0x48);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("ADS1X15_LIB_VERSION: ");
  Serial.println(ADS1X15_LIB_VERSION);

  Wire.begin();

  ADS.begin();

  Serial.println("\nTEST GAIN");
  int gain = 16;
  for (int i = 0; i < 6; i++)
  {
    Serial.print(gain);
    ADS.setGain(gain);
    if (ADS.getGain() == gain) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  
  Serial.println("\nTEST DATARATE");
  for (int i = 0; i < 7; i++)
  {
    Serial.print(i);
    ADS.setDataRate(i);
    if (ADS.getDataRate() == i) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  Serial.println("\nTEST MODE");
  for (int i = 0; i < 2; i++)
  {
    Serial.print(i);
    ADS.setMode(i);
    if (ADS.getMode() == i) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  Serial.println("\nTEST COMP MODE");
  for (int i = 0; i < 2; i++)
  {
    Serial.print(i);
    ADS.setComparatorMode(i);
    if (ADS.getComparatorMode() == i) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  Serial.println("\nTEST COMP POLARITY");
  for (int i = 0; i < 2; i++)
  {
    Serial.print(i);
    ADS.setComparatorPolarity(i);
    if (ADS.getComparatorPolarity() == i) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  Serial.println("\nTEST COMP LATCH");
  for (int i = 0; i < 2; i++)
  {
    Serial.print(i);
    ADS.setComparatorLatch(i);
    if (ADS.getComparatorLatch() == i) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  Serial.println("\nTEST COMP QUECONVERT");
  for (int i = 0; i < 2; i++)
  {
    Serial.print(i);
    ADS.setComparatorQueConvert(i);
    if (ADS.getComparatorQueConvert() == i) Serial.println("\tOK");
    else Serial.println("\tFAIL");
  }

  Serial.println("\ndone...");
}

void loop()
{
}

//  -- END OF FILE --
```

RobTillaart, 2024-08-20:
Note to myself: check if unit tests can be added

RobTillaart, 2024-08-20:
@HineSite
Created a develop branch with a preliminary fix, 
If you have time, please verify if it works as expected.

If the readme.md needs additional information on latch/polarity topic feel free to provide such.
(now back to my other tasks)

RobTillaart, 2024-08-20:
@HineSite 
Build of develop branch is stable, created a PR for release 0.5.0

RobTillaart, 2024-08-20:
Thanks for running the testcode
That confirms the bug and shows there are no other inverted flags (as far as test goes).

closed 2024-08-21 by RobTillaart

RobTillaart, 2024-08-21:
@HineSite 
Merged PR into master,
Again, thanks for reporting this issue, appreciated!

commit be3f8bc1 (2024-08-20): Fix #80, setComparatorPolarity() et al.
commit 760f89f6 (2024-08-21): Fix #80, comparator polarity and latch (#81)

- Fix #80, setComparatorPolarity() and setComparatorLatch() inverting.
- add test example to test parameters.
- add unit tests to test parameters.