RobTillaart/ADS1X15 #85: This is my code : (After many tests, I found that it doesn't work after the 0.3.9 version.

YLC35, 2024-10-29:
This is my code : (After many tests, I found that it doesn't work after the 0.3.9 version.
Thanks for your reactivity   ;-))

```cpp
// test lecture ADS1115 (I2C) SUR ESP32
// BROCHAGE :
//  pin 21   ==>  SDA ADS1115 (I2C)
//  pin 22   ==>  SCL ADS1115 (I2C)

#include "ADS1X15.h"         // lib from Rob Tillaart  ADS1X15 v 0.3.9 (incluse dans le dossier) 
ADS1115 ADS(0x48);           // instance for the ADS1115 
int16_t valA0 = 0;           // valeur brute lue sur A0 (15 bits)
int16_t valA1 = 0;           // valeur brute lue sur A1 (15 bits)
int16_t valA2 = 0;           // valeur brute lue sur A2 (15 bits)
int16_t valA3 = 0;           // valeur brute lue sur 32 (15 bits)
int16_t difA2A3 = 0;         // valeur brute lue sur différence A2 - A3 (15 bits) 

void setup() {
  Serial.begin(115200);
  ADS.begin();                                        // démarrage ASD1115
  ADS.setDataRate(7);                                 // Taux de conversion rapide
  ADS.setMode(1);                                     // mesures à la demande
  ADS.readADC(0);                                     // lecture vide pour prise en compte des paramètres
  ADS.setGain(0);                                     // plage de lecture 0 à 6v.

}

void loop() {
   valA0 = ADS.readADC(0);              // lecture valeur brute A0 en 15 bits
   delay(100);
   valA1 = ADS.readADC(1);              // lecture valeur brute A1 en 15 bits
   delay(100);
   valA2 = ADS.readADC(2);              // lecture valeur brute A2 en 15 bits
   delay(100);
   valA3 = ADS.readADC(3);              // lecture valeur brute A3 en 15 bits 
   delay(100);
   difA2A3 = abs(ADS.readADC_Differential_2_3());  // différentiel de tension entre A2 et A3

   Serial.print(valA0);
   Serial.print("  ");
   Serial.print(valA1);
   Serial.print("  ");
   Serial.print(valA2);
   Serial.print("  ");
   Serial.print(valA3);
   Serial.print("  ");
   Serial.println(difA2A3);
   delay(500);
}
```

(updated code tags for syntax highlighting)

_Originally posted by @YLC35 in https://github.com/RobTillaart/ADS1X15/issues/78#issuecomment-2443816729_

YLC35, 2024-10-29:
I finish to found how create a new issue.  :-))
I made the test of your example ADS.read but it doesn't neither work .
I made these tests on a ESP32 DEV kit v1 and an ESP32 Dev module. 
I called a developer friend and he did the same tests with the same problem.

YLC35, 2024-10-29:
Thanks you !  I can specify that I am running these tests on an ADS1115.

RobTillaart, 2024-10-29:
Got it, output of your sketch modified

```
3166  2813  2800  28 2799  2812  2847  0
3163  2810  2807  2824  2
2804  2795  2807  2822  -1
2811  2783  2810  2815  1
2805  2812  2808  2823  2
2777  2800  2812  2819  -1
2770  2813  2810  2810  0
2791  2794  2812  2818  1
2801  2809  2812  2831  1
2795  2810  2811  2849  1
2793  2812  2809  2819  3
2801  2810  2809  2831  -1
2779  2811  2806  2833  0
```

RobTillaart, 2024-10-29:
From the readme.md file

### 0.4.0 Breaking change

Version 0.4.0 introduced a breaking change. You cannot set the pins in begin() any more. This reduces the dependency of processor dependent Wire / I2C implementations. The user has to call Wire.begin() and can optionally set the I2C pins before calling begin().

----

This change has been introduced as different processors use different ways to define the I2C pins. 
As the pre-0.4.0 versions did try to work for all those processors the code got (censored).

So the decision was made to use proper dependency injection instead, which caused  a breaking change for a leap in portability.

To solve your code add ```Wire.begin()``` before ```ADS.begin()``` in setup().

```cpp
void setup() {
  Serial.begin(115200);

  Wire.begin();
  ADS.begin();                                        // démarrage ASD1115

  ADS.setDataRate(7);                                 // Taux de conversion rapide
  ADS.setMode(1);                                     // mesures à la demande
  ADS.readADC(0);                                     // lecture vide pour prise en compte des paramètres
  ADS.setGain(0);                                     // plage de lecture 0 à 6v.

}
```

YLC35, 2024-10-29:
Thank you very much. It actually works.
Too bad, because I had not seen this change of use which has not been highlighted enough in my opinion. 
Best regards YLC

RobTillaart, 2024-10-29:
>  I had not seen this change of use which has not been highlighted enough in my opinion.

It is in the readme under **breaking changes**, maybe I should have a separate lisezMoi? 😉
Anyway good to hear the problem is also solved at your side.