RobTillaart/ADS1X15 #86: ADS1118

Fluxanode, 2024-11-21:
Have you considered creating a library for the ADS1118?

RobTillaart, 2024-11-21:
Yes, actually I have, 
(pause)

it is on my "bucketlist" since 2022 dec 6 (looked it up) but never needed one in a project as the ADS1115 filled my needs.
Furthermore, it was not possible to find a decent breakout so I could test with UNO and ESP32. 
Don't know the current availability, do you know an ADS1118 breakout supplier? (Europe preferred).

So in short current status of my ADS1118 library is that it is somewhere on a list with 50++ other library-ideas. 

Do you need it on short term?
Do you have special requirements for an ADS1118 library? 
Functions that are not in the ADS1X15?

(You could sponsor the development, estimate it takes somewhere between 10-15 hours excl hardware testing)

Fluxanode, 2024-11-21:
I would be glad to do any testing for you, i have a test bed.

RobTillaart, 2024-11-21:
mouser says  181,34 euro.
expensive imho

> What is sponsoring?

Sponsoring is that you donate money so my work hours are paid for and I can buy the hardware needed to test. The latter is an option if you do not need testing, or do the testing yourself.

Note: The library will be MIT licensed so everyone may use it, no warranties. Just like all my Arduino libraries.

You can also donate money because you use one or more of my (other) libraries for free and you appreciate them and you want to support my work. But it is not mandatory to sponsor, your decision.

Fluxanode, 2024-11-21:
Yes it is expensive. I put the ic on a schmart board and setup on a breadboard.  If you do the code I'll do the testing, it should be pretty much your same library code but using the spi lib?  I understand the licensing.  Let me know what you do.

Thanks,

Chuck Weaver

RobTillaart, 2024-11-21:
Hi Chuck,

Have you seen - https://github.com/denkitronik/ADS1118 ?
Seems a stable and tested library for the ADS1118, 
Drawback is 2 open PR's and 11 open issues, oldest about 5 yrs, same age as last commit .

That said it might still meet your needs.

> Let me know what you do.

I understand your proposal code / testing, but there is no reason/incentive for me to write it as I have no project that needs it. 
Not now, and given its price not in the nearby future, sorry.
So ADS1118 stays on my todo list and eventually I write it in the future when time permits.

Fluxanode, 2024-11-21:
Ok thanks for your time.

RobTillaart, 2024-11-21:
You're welcome,

Also found this link which uses that library
- https://electropeak.com/learn/interfacing-ads1118-16-bit-adc-module-with-arduino/

and I moved it some places in my "bucketlist" 

Please let me know if the other lib meet your needs as that might be informative for other people
Thanks,
Rob

Fluxanode, 2024-11-21:
It seems to work ok, but when I was having what i think was a spi com issue (still don't know for sure) there is no error info reporting bad data, it just sent junk data.  Basically when it's working it's fine, but when something is off there is no way to troubleshoot.  Anyway I can live with it and maybe on the next design I'll use the ADS1115 but I'll need to add a board temperature sensor and read it with a lower resolution ADC on the teensy.

Thanks

closed 2024-11-22 by RobTillaart