RobTillaart/ADS1X15 #87: Observation on fast differential example

Dario503, 2025-01-25:
Rob, thanks for an excellent library. Recently I designed a system with 2 ADS1115 connected to an ESP32-S3. I used differential mode to capture multiple cycles of an AC waveform from a Current Transformer.  I found it necessary to set the interrupt to trigger on the falling edge of the Alert/Ready pin output.  Here is how I have things setup:

    pinMode(ADC1_RDY_PIN, INPUT);
    pinMode(ADC2_RDY_PIN, INPUT);
    attachInterrupt(digitalPinToInterrupt(ADC1_RDY_PIN), adc1Ready, FALLING);
    attachInterrupt(digitalPinToInterrupt(ADC2_RDY_PIN), adc2Ready, FALLING);
    adc1.setGain(2);                         // ±2.048V
    adc1.setDataRate(7);                     // 860 SPS
    adc1.setMode(0);                         // Continuous-conversion mode
    adc1.setComparatorThresholdHigh(0x8000); // MSB = 1
    adc1.setComparatorThresholdLow(0x0000);  // MSB = 0
    adc1.setComparatorPolarity(0);           // Active low (default behavior)
    adc1.setComparatorQueConvert(0);         // Keep ALERT/RDY pin enabled

You will notice in the attached scope pictures, that with the setup above the pulses are positive going. I found it necessary to set the interrupt to the falling edge of the pulse to get the most accurate readings (I am calculating TRMS). That seems to go contrary to what you state in your documentation. In the TI AD1115 data sheet, it states the data is ready on the falling edge of the positive going 8uS pulse. Anyway, you might want to revisit this, and I am happy to share additional info.

[image]
[image]
[image]

[image]

RobTillaart, 2025-01-25:
Thanks for this info, appreciated.
The datasheet is leading so I will check this coming week. 
Can you say which version of the datasheet you used?

Also great you added the scope-shots!👍

Dario503, 2025-01-26:
Rob, the TI data sheet was revised December 2024 and can be found here:

[https://www.ti.com/lit/ds/symlink/ads1115.pdf?ts=1737794024019&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FADS1115](https://www.ti.com/lit/ds/symlink/ads1115.pdf?ts=1737794024019&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FADS1115)

See page 17 for diagram. Text is still a bit nebulous, "The COMP_POL bit continues to function as expected".

Daryl

RobTillaart, 2025-01-26:
@Dario503 

Your remark is about this table in the readme.md if I understand correctly.

#### ALERT RDY table

|  MODE          |  COMP_POL  |  IDLE  |  START    |  CONVERT  |   READY   |
|:---------------|:-----------|:-------|:----------|:----------|:----------|
| 0 = continuous |  0 = LOW   |  HIGH  |  FALLING  |  LOW      |  RISING   |
| 0 = continuous |  1 = HIGH  |  LOW   |  RISING   |  HIGH     |  FALLING  |
| 1 = single     |  0 = LOW   |  HIGH  |  FALLING  |  LOW      |  RISING   |
| 1 = single     |  1 = HIGH  |  LOW   |  RISING   |  HIGH     |  FALLING  |

See issue #76 for some screenshots.

RobTillaart, 2025-01-26:
@Dario503 

(downloaded the latest datasheets, thanks for the link)

Your screen shots seems in line with the discussion in #76 , 
in fact your setup is missing there as screenshot (so thanks again).

From datasheet
_The ADS111x provide an approximately 8μs conversion-ready pulse on the ALERT/RDY pin at the end of each conversion in continuous-conversion mode_

Reading that section again and again and again and thinking out loud (always dangerous ;)

"provides a pulse at the end of conversion", that sounds to me as the pulse starts when the conversion is done.
So both edges should work to trigger fetching the result. (see hypothesis below why not).

>  I found it necessary to set the interrupt to the falling edge of the pulse to get the most accurate readings

What "effect" do you see in your measurements when using the rising edge? 
Can you explain in more detail how the data is better / more accurate?

### Hypothesis

An hypothesis is that at the start of the pulse, the result of the measurement is copied to the output register. 
And this copy action is ready at the end of the pulse, and new conversion starts.
(this is just a thought to explain more stable readings at the end of the pulse)

Dario503, 2025-01-26:
Excellent question: I have two calibrated TRMS AC current meters in series with the load. When I set the interrupt to the beginning of the pulse, the output of the software TRMS calculation tends to be inaccurate (+/- 20%) and wanders dramatically from reading to reading. When I set the interrupt to the end of the pulse, I get much better accuracy (+/- 2%) and stable results.

Yes. I have only tested with COMP_POL 0 (see register settings), but it appears the table may not follow actual.

I suspect during the 8uS pulse, conversion is occurring. The only thing alluding to that is the timing diagram I sent previously where it shows "Conversion Ready" at the end of the pulse.

RobTillaart, 2025-01-26:
Note: currently all my (interrupt) example sketches use RISING edge and none the FALLING edge.
I look into their code to see if there is a comment or so.

RobTillaart, 2025-01-26:
This is the text I propose to add to the readme.md, any additions?

_The examples of this library all use the RISING edge for the interrupt detection of the ALERT / RDY pin. In https://github.com/RobTillaart/ADS1X15/issues/87 it is observed that the FALLING edge gave far better results for the application used (determine True RMS) although this could not be explained. Thus changing the edge to FALLING might improve your measurements too. If anybody could explain the observed effect, please let me know._

Dario503, 2025-01-27:
I plan to do some more research and testing this week. Your text sounds good. While they don't explain it, the diagram in the latest data sheet does indicate that the 'Conversion Ready' is at the end of the pulse. I will test setting COMP_POL to 1, if it inverts the alert/rdy signal then your examples will be correct. :-)

[image]

RobTillaart, 2025-01-27:
Added a reference to the picture too.

RobTillaart, 2025-02-04:
@Dario503 

New insights?

Dario503, 2025-02-04:
@RobTillaart sorry, got side tracked on another project. Just bringing it up now and will advise.

Dario503, 2025-02-05:
I would submit based on the [Rev. D data sheet Figure 7-8](https://www.ti.com/lit/ds/symlink/ads1115.pdf?ts=1738663357472) that you might choose to update this part of the main text:

In short:

if COMP_POL = 0,

~~a FALLING edge indicates conversion start.~~
a LOW level indicates converting.
a RISING edge indicates conversion *completing*.
*a FALLING edge indicates conversion ready.*
~~a HIGH level indicates idle.~~

if COMP_POL = 1,

~~a RISING edge indicates conversion start.~~
a HIGH level indicates converting.
a FALLING edge indicates conversion *completing*.
*a RISING edge indicates conversion ready.*
~~a LOW level indicates idle.~~

*Don't forget, ALERT/RDY requires a pull-up resistor as it is an open-drain output.*

RobTillaart, 2025-02-05:
Thanks for all testing and the screen shots, really appreciated!
So just remove a few lines to get better documentation, I like that 😉

Will look into it asap

Dario503, 2025-02-05:
Anecdotally, I added some debounce to my interrupt handler and noticed it mitigates the issue that started this conversation (setting IRQ trigger on leading versus trailing edge of ALERT/RDY).  I surmise that somewhere within the 8us pulse the data is actually valid. When I get this project wrapped up, I'll go back and revisit and provide you with code that demonstrates the errant behavior. Also, perhaps there is someone at TI who might have the details as well (seeing they are still updating the datasheet). Thanks!

RobTillaart, 2025-02-05:
@Dario503 

Updated the readme.md in the develop branch, also updated the table above the section you referred to.

Dario503, 2025-02-05:
Rob, it looks good, thanks! 

Question for you, in single-ended mode with an ADS1115 set for data rate 7, you get 860 SPS. I am using both differential pairs in each 1115. It uses a mux to switch between them. Is it reasonable to conclude the effective sample rate is 430 SPS when using both differential pairs on the chip? The data-sheet doesn't specifically say this. It is important to me as I want to make sure I capture x full cycles of an AC waveform.

RobTillaart, 2025-02-09:
@Dario503 

Hi Daryl,

Is there anything you would add to the PR at the moment, otherwise I will merge it asap.

R.

Dario503, 2025-02-10:
Hi Rob,

Good to go. Thanks.

D

ejdp62, 2025-06-19:
Hello Rob, I have been busy in the past few weeks to get some conversions out of the ADS1115, with some success.
This is a response to your request in the readme: "If anybody can explain the observed effect, please let me know."

As the ALERT/RDY pin is open collector, the rising edges of the output signal and the falling edges of the output signal are quite different! All the scope pictures show this difference. The falling edges are much more clearly defined, because the signal is _actively_ pulled low, whereas it is only _passively_ pulled high. And that is true _regardless_ of the polarity chosen, as it is a consequence of the output cell in the chip (open collector driver), and not of the signal that the I/O cell is passing to the ready pin. I think this does explain all observations made.

Now that the data sheet has been updated to explicitly show that the conversion is ready on the _second_  slope of the pulse,
it is clear what to do best: set the polarity such that that second slope is _falling_ (as that is the superior direction), and use that one to trigger the interrupt. It may be appropriate to adapt the code in the library to implement only that  choice, as it will indeed give the best results.

ejdp62, 2025-06-21:
You are welcome, and please let me know if there are still some questions,
for example in the design and application of open collector output drivers.
(This is essential in understanding the physical layer of the I2C bus.)

Cheers, Evert-Jan

commit cbee197f (2025-02-11): fix #87, update readme.md about ALERT pulse and polarity (#88)

- Add note about FALLING edge when using interrupts, see #87
- update comments in some examples
- minor edits