RobTillaart/ADS1X15 #89: putting the ads1115 to sleep

arifainchtein, 2025-02-20:
Hi Robert,
i want to put the ADS1115 to sleep right before i put the esp32 to deep sleep.  How do i do that?
thanks

RobTillaart, 2025-02-20:
Thanks for the question,
First thing that comes in my mind is to set the device to single mode as continuous mode uses more energy.
Need to check the datasheet

RobTillaart, 2025-02-20:
@arifainchtein 

From the datasheet 

### Section 7.1

_The ADS111x have two available conversion modes: single-shot and continuous-conversion. In **single-shot**
mode, the ADC performs one conversion of the input signal upon request, stores the conversion value to an
internal conversion register, and then enters a **power-down** state. This mode is intended to provide significant
power savings in systems that only require periodic conversions or when there are long idle periods between
conversions._

### Section 7.4.1

_The ADS111x enter a power-down state after completion of the reset process._

Note: this is not the library **reset()** function.

-----

The library has no **sleep()** function and the **reset()** just resets parameters to default values.
So at the moment the only way to enter sleep mode is to set the mode to SINGLE-SHOT and ignore the reading.

So if you are using SINGLE SHOT, the ADC is already in sleep mode in between, otherwise you have to set this.

Does this answer your question?

RobTillaart, 2025-03-07:
As there is no sleep mode I close the issue
Feel free to reopen it if needed.

closed 2025-03-07 by RobTillaart