RobTillaart/ADS1X15 #91: Definitions of LATCH and NOLATCH seem wrong.

HenkHoldijk, 2025-06-20:
Hi,

Looking at ADS1X15.h the following 2 defines are available:
#define ADS1x15_COMP_POL_LATCH            0x00
#define ADS1x15_COMP_POL_NOLATCH          0x01

Looking at the Ti Datasheet (SBAS444D –MAY 2009–REVISED JANUARY 2018) page 29 this seems to be incorrect.
COMP_LAT "0" => Nonlatching comparator
COMP_LAT "1" => Latching comparator

At the same time, there is no inversion of the bits while using them in the library.

Cheers, Henk

RobTillaart, 2025-06-20:
Hi Henk, thanks for reporting, 
Topic is related to discussion in #80 
A quick analysis shows that these defines are not used in the library.

```cpp
void ADS1X15::setComparatorLatch(uint8_t latch)
{
  _compLatch = latch == 0 ? 0 : 1;
}
```

This code set the flag to 
- 0 when the parameter == 0 == false == NOLATCH and 
- 1 when the parameter != 0 == true == LATCH

So this function seems in line with your remark. 

Have some other tasks to do, and I will fix this issue asap.

HenkHoldijk, 2025-06-23:
B.t.w. This is what I use them for:

float ADS1115Handler::gainToFullScaleRange(uint8_t dGain)
{
  float fFSRange;

  switch (dGain)
  {
    case ADS1X15_GAIN_6144MV :
      fFSRange = ADS1115_GAIN_6144MV_FSRANGE_V;
    break;
    case ADS1X15_GAIN_4096MV :
      fFSRange = ADS1115_GAIN_4096MV_FSRANGE_V;
    break;
    case ADS1X15_GAIN_2048MV :
      fFSRange = ADS1115_GAIN_2048MV_FSRANGE_V;
    break;
    case ADS1X15_GAIN_1024MV :
      fFSRange = ADS1115_GAIN_1024MV_FSRANGE_V;
    break;
    case ADS1X15_GAIN_0512MV :
      fFSRange = ADS1115_GAIN_0512MV_FSRANGE_V;
    break;
    case ADS1X15_GAIN_0256MV :
      fFSRange = ADS1115_GAIN_0256MV_FSRANGE_V;
    break;
    default:
      fFSRange = -1.0;
  };

  return fFSRange;
}

and 

This small snippet:

dInt16 = static_cast<int16_t>(ADS1115_MAX_REG_VALUE * (fFloat / fFSRange));

For converting a float to a comparator threshold.

RobTillaart, 2025-07-07:
@HenkHoldijk 

> // ADS1115 Voltage Range Full Scale
#define ADS1115_GAIN_6144MV_FSRANGE_V   ( 6.144 )
etc

Should be 

```cpp
// ADS1x15 Voltage Range Full Scale
#define ADS1x15_GAIN_6144MV_FSRANGE_V   ( 6.144 )
#define ADS1x15_GAIN_4096MV_FSRANGE_V   ( 4.096 )
#define ADS1x15_GAIN_2048MV_FSRANGE_V   ( 2.048 )
#define ADS1x15_GAIN_1024MV_FSRANGE_V   ( 1.024 )
#define ADS1x15_GAIN_0512MV_FSRANGE_V   ( 0.512 )
#define ADS1x15_GAIN_0256MV_FSRANGE_V   ( 0.256 )
```

as it need to work for the 101x series too

Then add a new function **float getFullScaleRange()** that returns the full scale of the actual set gain.

```cpp
uint8_t  getGain(); 

float ADS1X15::getFullScaleRange()
{
  if (!(_config & ADS_CONF_GAIN)) return 0.0;
  switch (_gain)
  {
    case ADS1X15_PGA_6_144V: return ADS1x15_GAIN_6144MV_FSRANGE_V;
    case ADS1X15_PGA_4_096V: return ADS1x15_GAIN_4096MV_FSRANGE_V;
    case ADS1X15_PGA_2_048V: return ADS1x15_GAIN_2048MV_FSRANGE_V;
    case ADS1X15_PGA_1_024V: return ADS1x15_GAIN_1024MV_FSRANGE_V;
    case ADS1X15_PGA_0_512V: return ADS1x15_GAIN_0512MV_FSRANGE_V;
    case ADS1X15_PGA_0_256V: return ADS1x15_GAIN_0256MV_FSRANGE_V;
  }
  _error = ADS1X15_INVALID_GAIN;
  return _error;
}
```

RobTillaart, 2025-07-07:
```cpp
float ADS1X15::getFullScaleRange()
```
is implemented in

float    getMaxVoltage();

Just redo the implementation with the defines..

RobTillaart, 2025-07-07:
Time is up, start build :)

RobTillaart, 2025-07-08:
@HenkHoldijk 

Added some final comments and released 0.5.3 

Thanks again for your constructive ideas!

commit 29e919e3 (2025-06-22): fix #91, parameter constants
closed 2025-07-08 by RobTillaart