RobTillaart/ADS1X15 #93: Different gains for different channels

TechyGuy17, 2025-07-19:
Hello! Using your lib for a project where I need to messure 2 channels. One is quite low voltage and the other is about 5v (messuring VCC for calculations). Is it possible to have different gains for the 2 channels to get the precision for the low voltage?

RobTillaart, 2025-07-20:
Thanks for the question,
The device has one gain setting for all channels. This implies that when you change between low and high voltages you should set the appropriate gain. 
This takes extra time.

A hardware solution might be simpler. 
Add a voltage divider for the 5 Volt channel with such ratio that it uses the same gain as the low voltage channel. You need to add the (exact) ratio in your code to compensate of course. This hardware solution will probably be faster than the software solution.

As I do not know your project requirements I don't know which will be preferred.

TechyGuy17, 2025-07-20:
Thanks for the answer 😄 I am using the ADC to measure the ohm of a RTD
sensor. So the low voltage is actually a voltage divider with a very exact
resistor. Sadly speed is very important and getting more 0.001% resistors
will be very expensive :/ So I guess this will be good enough as it is
hehehe

Thank you!

RobTillaart, 2025-07-20:
If speed is important, I would consider a second ADC. Then you can optimize both

lynx985, 2025-07-20:
considering the price of an ADSxxx which starts at around 1$ for the 12 bit version it's hard to believe that 10 cents for two 0.1% resistors is too much

regarding your question: it is possible to read it non blocking with this library, make a request for a measurement and wait till it's done (or use the RDY pin to check if it's ready) and then read it out and request a new measurement. 
When the I2C speed is set to 1 MHz it takes less than 70us to read the date, so that would be the minimum blocking time

lynx985, 2025-07-20:
ah 0.01% that's different...

I tried to measure it, I don't understand how it can be that low, but I measure around 10 us @100kHz SCL
But that doesn't make sense, that would be only one clock cycle

A read or a request at 600 kHz is 100 us, so I would expect something along those lines...

RobTillaart, 2025-07-21:
> I am using it in a non blocking fashion right now, the question is just how much extra processing time/blocked time changing the gain adds. But feels like it would be quite little

mmm, should be easy to write a minimal test sketch to measure the extra time needed.

RobTillaart, 2025-07-27:
@TechyGuy17 

> So I guess this will be good enough as it is

As I assume your question is answered I close the issue

closed 2025-07-27 by RobTillaart