RobTillaart/ADS1X15 #94: current loss through ALERT pull-up

mbmorrissey, 2025-07-31:
Hi,

I find that when using the basic example ADS_read sketch, power consumption (Adafruit unit with power LED disabled) is about 330 uA (powered at 3.3V, so suspiciously close to what I guess we'd expect through a pullup resistor).

I think that this is the pullup on the ALERT pin, and when I do the following before sleeping, I end up with a current consumption from the ADS1115 unit of 10.8 uA.

```
  // read config register
  Wire.beginTransmission(0x48);  
  Wire.write(0b00000001); 
  Wire.endTransmission();
  Wire.requestFrom(0x48,2);
  uint16_t value = Wire.read() << 8;
  value += Wire.read();
  Wire.endTransmission(); 
  Serial.print("  "); Serial.println(value,BIN);

  // set COMP_POL bit to zero
  uint16_t mask = 0xFF << 8;
  mask += 0b11110111;
  uint16_t newval = value & mask;

  // write back to config 
  Wire.beginTransmission(0x48);  
  Wire.write(0b00000001); 
  Wire.write(newval >> 8);
  Wire.write(newval & 0xFF);
  Wire.endTransmission();
```

Am I missing a better way to use the library to avoid this current leakage?  Is there possibly an opportunity here for the library to turn off the alert before sleeping when in single-shot mode?

Thanks,

Michael

RobTillaart, 2025-07-31:
@mbmorrissey 

Do you need to reset the comparator mode when waking up in your setup?
(Additional code ?)

mbmorrissey, 2025-07-31:
Hi Rob,

Thanks for looking into this so quickly!

I added your comparatorOff() function to the .h and .cpp files locally, and it seems to work great.

Yes, things seem to be working well without any extra configuration on wake up.  I hooked up a potentiometer to A0 and it seemed to read voltages correctly every few seconds between sleeps.  Sleeps are down to 10.8uA as before with my crude attempt.

My test code (not including the obvious additions to the .h and .cpp files based on your above post) and example serial output is below.

Michael

Test sketch:
```
#include "ADS1X15_test.h"
#include <Wire.h>
ADS1115 ADS(0x48);

void setup() {
  Serial.begin(9600);
  Wire.begin();
  ADS.begin();
}

void loop() {

  // following is as per example sketch
  int16_t val_0 = ADS.readADC(0);
  float f = ADS.toVoltage(1);  //  voltage factor
  Serial.print("Voltage conversion factor: "); Serial.println(f,8);
  Serial.println();
  Serial.print("Analog0: "); Serial.print(val_0); Serial.print('\t'); Serial.println(val_0 * f, 6);

  // what the config register is presently, change it, verify change
  print_config();
  ADS.comparatorOff();
  print_config();

  // go to sleep, check (yes, 10.8 uA)
  delay(5000);
  Serial.println();
  Serial.println();
  Serial.println();
}

// print ADS1115 config registers, as in original post
void print_config(){
  Wire.beginTransmission(0x48);  
  Wire.write(0b00000001); 
  Wire.endTransmission();
  Wire.requestFrom(0x48,2);
  uint16_t value = Wire.read() << 8;
  value += Wire.read();
  Wire.endTransmission(); 
  Serial.print("  "); Serial.println(value,BIN);
}
```

Example output of test sketch:
```
17:50:08.166 -> Voltage conversion factor: 0.00018751
17:50:08.210 -> 
17:50:08.210 -> Analog0:         17924	3.360852
17:50:08.245 ->   1100000110001011
17:50:08.245 ->   100000110000011
17:50:13.267 -> 
17:50:13.267 -> 
17:50:13.267 -> 
17:50:13.267 -> Voltage conversion factor: 0.00018751
17:50:13.305 -> 
17:50:13.305 -> Analog0:         12831	2.405886
17:50:13.343 ->   1100000110001011
17:50:13.343 ->   100000110000011
17:50:18.314 -> 
```

RobTillaart, 2025-08-06:
Thanks Michael for the extensive explanation with pictures.
Will release the new version asap.

RobTillaart, 2025-08-07:
Merged and released 0.5.4, should be available in the library manager later today.
Thanks again for the issue and the testing, appreciated!

commit b5ddb354 (2025-08-07): fix #94, add setComparatorOff() (#95)

- fix #94, add setComparatorOff() function. (experimental)
- update keywords.txt
- update readme.md.
closed 2025-08-07 by RobTillaart