RobTillaart/ADS1X15 #96: Is "_conversionDelay" needed when using alert pin and continuous mode for reading all 4 channel?

unalartan-code, 2025-08-10:
Hi, trying to get the maximum speed from polling all 4 channels on a ADS1115. I see that on line 489 in the ADS1X15.cpp there is a delay of 8 ms even when in continuous mode. 

My question is: Is this needed when polling continuously using the alert pin to change channels so the example ADS_continuous_4_channel?

You make a comment if no delay then some old data is possibly obtained. If we keep it in then the fastest we can poll all 4 channels 31 Hz if not additional overhead with the current implementation and roughly 25 Hz when I test it with some Serial printing.

If you can provide your thoughts that would be great.

RobTillaart, 2025-08-10:
Thanks for the question, 
Will try to answer this week, need to reread the code to recall the details. 

You are free to modify/ tune that delay and if it still works within your requirements its great.

unalartan-code, 2025-08-10:
Thanks and for sure I have modified it and seems to work. Intuitively it seems to make sense to remove it I just wasn't sure if that 8 ms was an internal delay for switching channels.

Thanks for the quick prompt and thanks for your code repos!

unalartan-code, 2025-08-10:
Yeah I was investigating that issue comments intensively but for some reason didn't connect things... I'm going to investigate more and get back to you on any issues that I encounter with it being removed or a low value. Striving for at least a stable 100 Hz solution reading all 4 inputs.

RobTillaart, 2025-08-10:
It might be an idea to change the delay() to a delayMicroseconds() so it can be tuned in finer timing steps.

RobTillaart, 2025-08-11:
@unalartan-code 

> Yeah I was investigating that issue comments intensively but for some reason didn't connect things... 

As far as I recall the delay is used to skip over one reading, 
If you switch channel in continuous mode the moment you get an ALERT the new conversion starts. 
This is the last channel.
The delay takes care that you are sure the conversion of the requested channel will be read. 

----

(thinking out loud section, BIG assumptions ahead)
So it could be that to maximize sampling, one need to request channel B before channel A is ready.

Then the ALERT will state channel A is ready and continues with channel B. 
The user reads channel A and request channel C

Then the ALERT will state channel B is ready and continues with channel C. 
The user reads channel B and request channel D

Then the ALERT will state channel C is ready and continues with channel D. 
The user reads channel C and request channel A

Then the ALERT will state channel D is ready and continues with channel A. 
The user reads channel D and request channel B
etc

To be tested with hardware ...

unalartan-code, 2025-08-11:
Hi Rob, you beat me to the answer. On TI website the swtiching of channels should be instant but like you mention it may be starting a new conversion on the old channel. So the delay makes sense but should be adjusted based on the desired SPS as right now it is hard coded at the 125 SPS. One could also check to make sure you see 2 RDY signals and use that value?

Surprisingly, I have an Arduino MKRZERO attached to an Adafruit ADS1115 and commenting out the delay I get reliable analog values (no jumping channels) but get ~420 SPS so a 100 Hz system which is ok for me if it is stable. 

Wild guessing here but maybe the comparator queue is compensating for this I don't know... I will report back on the stability of the system when I have my full setup wired (4 ADS1115 units polling all 4 channels as fast as possible...).

Thanks again.

RobTillaart, 2025-08-19:
> Surprisingly, I have an Arduino MKRZERO attached to an Adafruit ADS1115 and commenting out the delay I get reliable analog values (no jumping channels) but get ~420 SPS so a 100 Hz system which is ok for me if it is stable.

Possibly your code (in combination with the MKRZERO) triggers the sampling just fast enough to switch channels in time.
In my library I have to be compatible with slow devices (UNO R3 16 MHz).

As the problem seems to be understood / is solved I close the issue.
Feel free to reopen (or start a new) if needed.

closed 2025-08-19 by RobTillaart